ajfisher/node-pixel #5: Enable data pin other then pin 6

timmeador, 2015-02-05:
I am trying to run two instances of pixel strips on different arduino pins.  How hard would it be to enable the functionality to activate other data pins?

ajfisher, 2015-02-06:
PRs gratefully accepted. 😉

OK so this isn't trivial because of the bindings between firmata and the
adafruit library.

Basically what you need is multiple instances of the adafruit objects. This
has a memory implication as it's very heavy per pixel & strip.

As the JS library is managing the whole state of pixels etc, I was planning
on stripping this side back significantly to simply be a strip of memory
with each three bytes being the RGB components of each pixel.

Then you store a pin number and a pointer to the memory location and
allocate how much you need for each strip (that way you can have 1 strip of
80 & 2 of 40 in the same setup).

This is basically my next priority for the code once I'm back from vacation
next week.

xefcat, 2015-05-08:
First of all, thxs a lot for the library and the firmata firmware with NeoPixel support!

We are also trying to control multiple instances of NeoPixel strips on different arduino pins. We would love to try to help with the code, or at least testing...

ajfisher, 2015-05-18:
Pin changing pins is available in the branch I'll be pushing tonight. 

I need to get I2C support in place first and will come back to a basic implementation that allows for more than one strip. Note though that adding multiple strips SIGNIFICANTLY increases the firmata memory footprint size. 

Also, when the adafruit library latches the strip it blocks the main loop so it will slow down your performance on the firmata side.

Once I get the I2C side sorted this should be less of a problem though

ajfisher, 2015-05-24:
Changing pins is now available as part 526aa82c389c223287a9da8c9e0e21ad5fec7bb8 

Next step will be to add multi-strip support which I will bring open in a new thread.

ajfisher, 2015-05-24:
Closing this as this has broken out into an entire milestone.

@xiroi @timmeador If you'd be interested in helping out there are plenty of tasks to get on with on this one!

https://github.com/ajfisher/node-pixel/milestones/Multistrip%20support will comprise all of these tasks.

closed 2015-05-24 by ajfisher