ajfisher/node-pixel #6: Doesn't added to NPM registry

panayotov, 2015-03-24:
Hi,

When I try to install the package via NPM, I get the following error:

npm ERR! 404 Not Found: node-pixel
npm ERR! 404 
npm ERR! 404 'node-pixel' is not in the npm registry.
npm ERR! 404 You should bug the author to publish it (or use the name yourself!)
npm ERR! 404 It was specified as a dependency of 'speach'
npm ERR! 404 

It also does give me errors if I try to install it manually ( clone it from Github and install dependencies via NPM : )

ajfisher, 2015-03-24:
Sorry my bad - I haven't published to npm yet - as it's still a little
unstable. What errors are you getting if you git clone and npm install?

panayotov, 2015-03-24:
Hi, thank you for the fast reply,

This is what I get ( installing dependencies seems to be OK ) , but when I run the sample code I get this:

Error: Cannot find module 'node-pixel'
    at Function.Module._resolveFilename (module.js:322:15)
    at Function.Module._load (module.js:264:25)
    at Module.require (module.js:351:17)
    at require (module.js:370:17)
    at Object.<anonymous> (/Users/nxt/Desktop/rgb led/app.js:12:13)
    at Module._compile (module.js:446:26)
    at Object.Module._extensions..js (module.js:464:10)
    at Module.load (module.js:341:32)
    at Function.Module._load (module.js:296:12)
    at Function.Module.runMain (module.js:487:10)
NXT:rgb led nxt$

ajfisher, 2015-03-24:
> at Object. (/Users/nxt/Desktop/rgb led/app.js:12:13)

hmm seems like there might be a dependency path issue. you're running the
example from the root of your project yes ? Ie node
node_modules/examples/.... etc

ajfisher, 2015-04-22:
Did this get resolved??

closed 2015-05-18 by ajfisher