ajfisher/node-pixel #16: Add ESP8266 Controller

ajfisher, 2015-05-24:
Add a controller to be able to send messages to the ESP8266 using a dedicated firmware for this.

@gorhgorh Maybe something you want to pick up??

gorhgorh, 2015-05-24:
hey hey, I'm in berlin for a few day with no esp at hand but when i come back I'll be glad to have a shot at it

MatijaMartinec, 2016-09-29:
Hello, I was wondering if there was any progress being made for the ESP8266 board since trying to compile the firmware for the ESP8266 is futile (AVR/io.h and co. are missing as ESP8266 is not an AVR device). Thanks in advance.

ajfisher, 2016-09-29:
Slow progress at this point. The ESP* is a very capable device but it has
some quirks and additionally the iteration time of just build& flash is
quite time consuming.

Currently the standard node pixel firmware will build for esp* quite
happily as that is just C. What I need to get working is the DMA -> I2S
part so we can have decent frame rates on large numbers of pixels. My
knowledge on this is pretty sketchy though.

It's definitely seeing active but slow development at the moment. If you're
familiar with DMA / I2Sv then that's where I need help.

If you want something to just "work", then I've had very good success with
arduino running node pixel firmata joined to an esp8266 running a UDP wifi
to serial bridge works pretty well.

ajfisher, 2017-01-19:
Looks like I have a path forward with this given work I've done recently on ESP8266 and getting things working with MQTT etc. Need to rationalise some work done at https://github.com/ajfisher/espixel to back port that back to NP.

ajfisher, 2017-01-20:
Approach will be:

* Custom firmware that will be deployable on ESP8266
* Configurable pin for strip
* Custom length of strip
* Firmware specific interface protocols (SERIAL, UDP, MQTT) in order to accept messaging.
* Need to look at what the onboarding part to this looks like as well given some of the challenges I've seen with this side of things.

ScreamZ, 2023-11-20:
@ajfisher Is there any support today for ESP32 or ESP8266?

ajfisher, 2023-11-22:
There's not. It's feasible to cross compile firmata to work on ESP32/8266
but there's definitely some differences around how the underlying hardware
works which means things like how you run the stream of data isn't a direct
port for the library. Am very time poor at the moment to try and tackle
this however I'm very happy to discuss an approach for this if someone is
up for tackling it as I have got a sketch for this sitting on my backburner.