ajfisher/node-pixel #20: Add matrix support

ajfisher, 2015-05-24:
Add the ability to have a rectangular matrix of pixels and support this natively.

ajfisher, 2015-12-03:
In multipin multiple strips will be available as well as single strip

Matrix support need to support (at minimum):

* RC, CR order 
* 3 dimensional matrices
* Take single strip and split into a matrix NxM (eg 1x25 single strip can become 5x5 square)

VictorioBerra, 2018-08-15:
> Take single strip and split into a matrix NxM (eg 1x25 single strip can become 5x5 square)

This is exactly what I came here to find out if this library can do. I know its been a few years now, do we foresee this ever actually happening?

ajfisher, 2018-08-15:
Unfortunately my time is pretty limited at the moment but PRs are
gratefully received. I can help guide you through what's needed to be done
too.

jakepearse, 2018-12-11:
What is rc, cr order?

ajfisher, 2018-12-11:
Row-column or column-row order. It's used to designate how you apply a 2D
matrix to a 1D array.

jakepearse, 2019-02-17:
My son and I have been playing with this requirement.
Rather than messing about with a bunch of arrays, we used functions to re-map the addresses of pixels in a serpentine array.
Because node-pixel uses functions for addressing rather than presenting strips as 'array-like'.
It wouldn't take much to add cr/rc functionality, I guess.
Although it would perhaps add the need for an argument/options style object, using minor/major axis rather than column/row.
Which would probably better than a bunch of unnamed positional arguments the way we did it.
Anyway its all in a nice-and-small PR :)