ajfisher/node-pixel #24: I2C baudrate issues with Nano backpacks

frxnz, 2015-05-27:
This issue cropped up with Raspberry Pi, and may be specific to that board. Essentially it defaults to a baudrate (100000) that is too fast for the Nano. Current workaround is to set the baudrate to 10000 in `boot/config.txt`.

Is there any way to work around this without modifying the board (Pi) config?

cc @fivdi

frxnz, 2015-05-27:
Forgot to attach this. The issue is linked to a bug with Raspberry Pi's I2C implementation: http://www.advamation.com/knowhow/raspberrypi/rpi-i2c-bug.html

fivdi, 2015-05-27:
Since this is related to a Broadcom BCM2835 (Pi 1) and BCM2836 (Pi 2) clock stretching hardware bug there's probably not much that can be done about it. The bug prevents I2C devices, the Nano in this case, from stretching the clock in ways that should be possible.

The results on other boards like the BeagleBone Black should be better.

ajfisher, 2015-06-03:
Let'ss park this for the moment but good that it's here for google as to where the issue is coming from.

closed 2015-06-03 by ajfisher