ajfisher/node-pixel #28: Add support for GRB Neopixels

frxnz, 2015-06-10:
The [through-hole LED Neopixels](http://www.adafruit.com/products/1734) use a GRB format instead of RGB. Should add an initialization param to make this work.

ajfisher, 2015-06-11:
So normal strip neopixels use the same format too. I haven't seen any WS2812B 's that don't. Having said that, other LEDs can take different forms so this is a good idea. The NP library handles this but it does it in a series of #define statements that get used at object instantiation.

I'll visit this as part of #15 as we need to work through a method of attaching those objects dynamically as needed rather than all at once so things like Pin, strip type, speed, colour order etc all need to be worked out to be configurable.

crayneogeo, 2015-06-11:
this would be a great enhancement for people using the adafruit single neopixels!

ajfisher, 2015-12-02:
This is open again because need to do:

* Document how to set this as part of the constructor for the strip.
* Needs to be done at the per strip level as part of the `Strip` options when specifying the pins and lengths.

ajfisher, 2015-12-05:
Prep done in bb8523bb55fbca63cbb6192523542b9fff1c29fb

commit bd0cf357 (2015-12-02): Pin specify and colour order. Fixes #47 Closes #28
commit d8ae06cf (2015-12-05): Provides proper strip handling. Closes #28 closes #14 closes #12
closed 2015-12-09 by ajfisher