ajfisher/node-pixel #29: NeoPixel ring gets "stuck" after running for 15 - 30 minutes

cylinderStudio, 2015-06-12:
I'm writing a program where light animations coincide with actions from a separate web application (signal sent via websocket), and a sequence of animations run in a loop in a resting or attract state. 

The applications starts and runs. The attract sequences run in their loop when the ```strip.on('ready')```. And the other animations run when commands come in via websocket.

But when I let the attract sequences loop run for anywhere between 15 and 30 minutes, the NeoPixel ring I'm using stops taking the ```strip.show()``` commands and the LEDs get stuck. It always seems to happen when all LEDs are on ```rgb(180,180,180)```.

I suppose it could be my NeoPixel ring, or my code (below). But the application doesn't quit. It keeps running with no errors.

```
var pixel = require('node-pixel');
var five = require('johnny-five');
var io = require('socket.io')(8000);

var opts = {};
opts.port = process.argv[2] || "";

var board = new five.Board(opts);
var strip = null;

var attractSequence = [];
var sequence_idx = 0;
var animationInterval;
var feedbackAnimations = {};

var NUM_PIXELS = 24;
var MILLIS = 35; 

board.on('ready', function() {

  strip = new pixel.Strip({
    data: 6,
    length: NUM_PIXELS,
    board: this,
    controller: 'FIRMATA',
  });

  attractSequence.push(
    // quads
    function() {
      console.log('quads');
      var quads = { quad1: 0, quad2: 0, quad3: 0, quad4: 0 };
      var p;
      var frame_count = 0;

      strip.color("black");

      function processQuad(start_pixel,quad) {
        for (var i=start_pixel; i < (start_pixel+6); i++) {
          p = strip.pixel(i);
          p.color("rgb("+quads[quad]+", "+quads[quad]+", "+quads[quad]+")");
        }
        quads[quad] = quads[quad]+12;
      }

      animationInterval = setInterval(function() {
        if (quads.quad1 < 180) { processQuad(0,"quad1"); strip.show(); }
        else if (quads.quad1 >= 180 && quads.quad2 < 180) { processQuad(6,"quad2"); strip.show(); }
        else if (quads.quad2 >= 180 && quads.quad3 < 180) { processQuad(12,"quad3"); strip.show(); }
        else if (quads.quad3 >= 180 && quads.quad4 < 180) { processQuad(18,"quad4"); strip.show(); } 

        frame_count++;
        if (frame_count >= (1000/MILLIS)*8) { feedbackAnimations.attract(); }
      }, MILLIS);
    }
  );

  attractSequence.push(
    // "breathe" or yo-yo fade
    function() {
      console.log('yo-yo');
      var current_color = 0;
      var mult = 1;
      var frame_count = 0;

      strip.color("black");

      animationInterval = setInterval(function() {
        strip.color("rgb("+current_color+", "+current_color+", "+current_color+")");

        if (current_color >= 180) { mult = -1; }
        if (current_color <= 0) { mult = 1; }

        current_color = current_color + (4 * mult);

        strip.show();

        frame_count++;
        if (frame_count >= (1000/MILLIS)*8) { feedbackAnimations.attract(); }
      }, MILLIS);
    }
  );

  feedbackAnimations.attract = function() {
    console.log('attract advance');
    if (animationInterval) clearInterval(animationInterval);
    attractSequence[sequence_idx]();
    if (sequence_idx+1 >= attractSequence.length) { sequence_idx = 0; }
    else { sequence_idx++; }
  }

  feedbackAnimations.fadeIn = function() {
    var current_color = 0;

    strip.color("black");

    animationInterval = setInterval(function() {
      strip.color("rgb("+current_color+", "+current_color+", "+current_color+")");
      current_color = current_color + 4;
      strip.show();
      if (current_color >= 180) clearInterval(animationInterval);
    }, MILLIS);
  };

  feedbackAnimations.fadeOut = function() {
    var current_color = 180;

    animationInterval = setInterval(function() {
      strip.color("rgb("+current_color+", "+current_color+", "+current_color+")");
      current_color = current_color - 4;
      if (current_color <= 0) {
        clearInterval(animationInterval);
        strip.color("black");
      }
      strip.show();
    }, MILLIS);
  };

  feedbackAnimations.spinner = function() {
    current_pos = 0;

    animationInterval = setInterval(function() {
      strip.color("rgb(180, 180, 180)");

      var p = strip.pixel(current_pos);
      p.color("black");
      if (current_pos+1 >= strip.stripLength()) { current_pos = 0; }
      else { current_pos++; }

      strip.show();
    }, MILLIS);
  };

  function resetStrip() {
    // clear strip & animation 
    if (animationInterval) clearInterval(animationInterval);
    strip.color("black");
  }

  strip.on('ready', function() {
    feedbackAnimations.attract();

    io.on('connection', function(socket) {
      socket.on('lights-attract', function(data) {
        resetStrip();
        feedbackAnimations.attract();
      });

      socket.on('lights-fade-in', function(data) {
        resetStrip();
        feedbackAnimations.fadeIn();
      });

      socket.on('lights-fade-out', function(data) {
        resetStrip();
        feedbackAnimations.fadeOut();
      });

      socket.on('lights-spinner', function(data) {
        resetStrip();
        feedbackAnimations.spinner();
      });

    });
  });

});
```

ajfisher, 2015-06-13:
Interesting. Do you have a version of this running without any web sockets
stuff I can try out? Want to try and replicate this with the simplest
application we can as there are quite a few moving pieces in this and at
least several possibilities as to where the problem is.

I've let the example demo run all day before without any problems but that
doesn't mean it's not that... It could be things stacking together.

cylinderStudio, 2015-06-15:
I think you're on to something. I don't have the same Mac available to me from my initial test, but I have the same Arduino Uno, same NeoPixel ring, same version of Node and Mac OS X on an older Macbook running the code minus the Socket.io code and it's all been running fine for three hours now.

cylinderStudio, 2015-06-17:
I'm closing this, because I'm running the original code again with the socket.io included on the first Mac I was testing with, and now it's running for hours at a time. I'll re-post in a new issue if it occurs again and I can tie it to something more specific.

closed 2015-06-17 by cylinderStudio

ajfisher, 2015-06-18:
Excellent news

sgtlambda, 2019-03-30:
Any chance we can reopen this @ajfisher ? I am experiencing the same issue,
The 60 LED ring was running fine for several minutes, then got "stuck" and now it doesn't light up anymore. I was running this very simple code:
```js
pixel = require("node-pixel");
five = require("johnny-five");

var board = new five.Board();
var strip = null;

board.on("ready", function() {

    strip = new pixel.Strip({
        board: this,
        controller: "FIRMATA",
        strips: [ {pin: 3, length: 60}, ], // this is preferred form for definition
        gamma: 2.8, // set to a gamma that works nicely for WS2812
    });

    strip.on("ready", function() {

        console.log('strip ready');

        strip.pixel(6).color([80, 0, 0]);
        strip.pixel(7).color([80, 0, 0]);
        
        strip.pixel(35).color([40, 40, 40]);
        strip.pixel(36).color([40, 40, 40]);

        strip.show();

        setInterval(() => {

            strip.shift(1, pixel.FORWARD, true);
            strip.show();

        }, 60);
    });
});
```

It still says "strip ready".

ajfisher, 2019-03-30:
Are you running this using I2C or node firmata?

sgtlambda, 2019-03-30:
Just figured out it was actually caused by a defective wire or breadboard. Sorry for bothering you.

ajfisher, 2019-04-03:
No worries at all. The number of times I've done exactly the same thing is
countless...