ajfisher/node-pixel #31: First 5 pixels staying red

pierceray, 2015-06-23:
I'm using an [AdaFruit NeoPixel Shield](https://www.adafruit.com/products/1430)

I tried to port over the rainbow function from the AdaFruit [strandtest example file](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/strandtest/strandtest.ino#L56).

So far it works, but with one small caveat, the first 5 pixels on the Arduino Shield stay red the entire time.  Every other pixel cycles the colors as expected.

I created a [gist with the example code](https://gist.github.com/pierceray/4aebbb877d0b64c83a29)

I added some `console.log` statements to illustrate the problem on the first pixel.

Truncated Output:
```
1435034348809 Device(s) /dev/cu.usbmodem1411
1435034348821 Connected /dev/cu.usbmodem1411
1435034352422 Repl Initialized
>> Board ready, lets add light
Strip ready, let's go
rainbow
rgb(255,0,0)
{ r: 255,
  g: 0,
  b: 0,
  hexcode: '#FF0000',
  color: 'red',
  rgb: [ 255, 0, 0 ] }
rgb(252,3,0)
{ r: 255,
  g: 0,
  b: 0,
  hexcode: '#FF0000',
  color: 'red',
  rgb: [ 255, 0, 0 ] }
rgb(249,6,0)
{ r: 255,
  g: 0,
  b: 0,
  hexcode: '#FF0000',
  color: 'red',
  rgb: [ 255, 0, 0 ] }
rgb(246,9,0)
{ r: 255,
  g: 0,
  b: 0,
  hexcode: '#FF0000',
  color: 'red',
  rgb: [ 255, 0, 0 ] }
rgb(243,12,0)
{ r: 255,
  g: 0,
  b: 0,
  hexcode: '#FF0000',
  color: 'red',
  rgb: [ 255, 0, 0 ] }
...
```

Notice that the value from the `strip.pixel(0).color()` values never change even though the rgb string being passed to it does.

pierceray, 2015-06-23:
Not that great of quality, but here's a video of what I'm seeing.  https://vimeo.com/131505601

ajfisher, 2015-06-23:
Can you try and do a couple of things:

1. Use the np-johnnyfive test script and make sure those pixels are
updating.

2. Specifically set the pixels not using the rainbow code but some hard
coded values

Under the rainbow wheel you're going to get them more or less "red" just
due to the amount of change they have however the .color(showcolor) should
update all of the elements of that value as it's going through.

Are the other pixels changing colour at all?

pierceray, 2015-06-23:
I've already shut everything down for tonight, but I did run those example scripts earlier and they seemed to work just fine.  I was also able to use the repl as well as hard code the pixel to set the colors to values other than red while I was trying to debug this.  I also tried using `strip.color('teal')` and all the pixels changed colors.

This video I linked the second comment shows what I'm seeing.  Every pixel except the first 5 are changing colors.

I'm hoping this is something in my ported code that's not right, but it seem like it would have manifested in all the pixels, not just the first 5.

ajfisher, 2015-06-23:
Okay - if you've had the np-j5 test running and done things like change the
whole strip different colours then you're totally good strip wise (these
pixels can "lift" off their pads which causes similar behaviour.

So that leads me to think it's some interaction between the colour wheel
code and something going on with node-pixel.

I am at work at the moment but have a 64pixel matrix I can use to try this
out on so I'll give that a crack tonight and see if I get a similar effect.

ajfisher, 2015-06-23:
So I've got to the bottom of this. There's a few things going on. Before I go into that, it's important to understand what is happening with your shield, firmata and your script. Apologies if you know this already - it might help someone else in the future too so it's here for Google as well.

For each pixel to light up you have to send a sequence of values down the wire to tell each pixel what to do. It's a FIFO queue in hardware. More importantly though, even if you want to change pixel 0 you will end up changing all 40 pixels because you HAVE to send the data through to the end. For this reason what we're doing is buffering the output. The frame isn't updated to the pixels until you run `strip.show()` - you can update as much as you like but as soon as you call that, you're committed - the arduino will block, the hardware will write to the strip and not do anything else until it's finished - this includes processing messages from firmata or over the wire.

So whenever you're sending messages it needs to get processed by firmata and before that Johnny-five to mediate those messages etc. Node pixel constructs the messages, johnny-five sends them, firmata interprets them into instructions and eventually they are then sent down that wire.

So why did you get a bunch of red pixels?

Basically you were sending messages so fast they kept getting clobbered and the red pixels were a side effect of overwriting your neopixel queue before it had a chance to finish the frame.

I'll focus on this specific area in your code as everything else is fine:

```javascript
function rainbow(delay){
  var foo = setInterval(function(){
//// ...
  }, 1000/delay);
}
``` 

As this was taken from the example which has a few moving pixels on a short strip, the Frames Per Second is 100. On reasonable numbers of neopixels this causes a LARGE number of messages to go across the wire (don't forget you have to push all those `pixel.color(rgb)` commands across the serial connection too.

```
for(var j = 0; j < 256; j++) {
  for(var i = 0; i < strip.stripLength(); i++) {
    showColor = Wheel( ( i+j ) & 255 );
    strip.pixel( i ).color( showColor );
    // code ...
  }
  strip.show();
}
```

As well as trying to execute the function 100 times a second, every time you call that function, the code will then try to update all 40 pixels and then push the frame down the wire 256 times during that function call. Which is a LOT. That's trying to execute 25000 frames per second. With all the messages that entails.

Why you're seeing the colour wheel animate though is that enough of the messages *are* getting through that the code is largely working, but you're also clobbering a lot and the number of red pixels you see is simply a function of how fast the arduino runs. A trinket would have fewer, a Mega might have more because they are clocked at different speeds - that might be an interesting way to benchmark chipsets / boards ;P

Final bit being the "not updating values".

```
/ Illustrates the problem
if (strip.pixel( 0 ).color().color === 'red'){
  console.log(showColor);
  console.log(strip.pixel( 0 ).color());
}
```

This was as simple as referencing pixel Zero not pixel(i) in the code.

To finish that part of, the adafruit code is trying to run as fast as physically possible because it's direct on the hardware and not doing anything else. In this instance that's not what you want to do.

A most basic version of the rainbow (fully static) would be:

```
function rainbow(delay){
  var showColor;
  for(var i = 0; i < strip.stripLength(); i++) {
    showColor = Wheel( ( i*256 / strip.stripLength() ) & 255 );
    strip.pixel( i ).color( showColor );
  }
  strip.show();
}
```

To get that to animate sort of like the adafruit one does then maybe something like this:

```
// All the top of your code is the same other than the fps is dropped to about 8fps

function rainbow(){
  var showColor;
  var cwi = 0; // colour wheel index (current position on colour wheel)
  var foo = setInterval(function(){
    if (++cwi > 255) cwi = 0;
    for(var i = 0; i < strip.stripLength(); i++) {
      showColor = Wheel( ( cwi+i ) & 255 );
       strip.pixel( i ).color( showColor );
    }
    strip.show();
  }, 1000/delay);
}

// all the rest is just the same too
```

In terms of frame rate you'll need to play with it. Between throughput over serial, firmata and pushing to my 64 pixel matrix I can only handle 8fps and my RX light on my arduino is solidly lit so I suspect that's most of the issue.

I tried it on another strip that was 16 pixels long and managed to easily get 30fps out of it so maybe with 40 you'd get about 20fps... 

Hopefully that sorts this issue out. I am planning on having some basic animation sequences available in the "backpack" version of this but that's yet to come.

closed 2015-06-23 by ajfisher

pierceray, 2015-06-24:
Thanks for the insightful explanation.

I've tried both implementations of the rainbow you listed.  They both work great.  The dynamic color one I was only able to get 13fps out of the 40 pixel NeoPixel shield.  At 14fps, the first pixel was flaky about half of the time with its color value.

Thank again for the explanation.

ajfisher, 2015-06-24:
Cool - and good to see you got some additional FPS out of that with the
shield. Even at 8fps on a 64 pixel display it was pretty smooth to be
fair... certainly smooth enough if you wanted some kind of lava-lamp type
effect...

There will be more optimisations on the protocol over time so this will
definitely get better. In this case what we really want is the ability to
shift the pixels one forward or one back along the entire strip array -
which is relatively straight forward and would cut down on the wire traffic
*significantly* - have added that to the enhancement list so I don't forget
about it as this is likely to be quite a common case.

This has been captured in #32

Also if you don't mind, I'd love to include the code we did as an example
as well?

pierceray, 2015-06-24:
I'll put in a pull request for it.

ajfisher, 2015-06-24:
Brilliant!