ajfisher/node-pixel #34: Installation Issue

moodybass, 2015-07-09:
Having an issue installing node-pixels.  I'm running Win7.  I'm able to Git Clone fine, however, I note that the following files are symbolic links:
Adafruit_NeoPixel.cpp
Adafruit_NeoPixel.h
WS2812.cpp
WS2812.h

I have the AdaFruit_* files from the NeoPixel library, however I can't seem to locate the correct WS2812.cpp or WS2812.h files - I have tried using the ones located here: https://github.com/cpldcpu/light_ws2812 however, I'm guessing they're not correct as I get a "'PIXEL_COMMAND' was not declared in this scope" error when I attempt to compile.

Any advice you could give would be greatly appreciated!

moodybass, 2015-07-09:
OK - bit more playing after posting this.  I found the correct files in a previous version:
https://github.com/ajfisher/node-pixel/blob/526aa82c389c223287a9da8c9e0e21ad5fec7bb8/firmware/libs/ws2812/ws2812.h

Wondering if there is an issue with the current version?

frxnz, 2015-07-09:
@moodybass All of the libs are still here: https://github.com/ajfisher/node-pixel/tree/master/firmware/libs

I'm not having any issues on OSX, but I can confirm the issue on Windows. My guess is that Windows doesn't like how the symlink paths are formatted, though I'm not familiar enough with symlinks in Windows to know what the fix is.

moodybass, 2015-07-09:
Ahhh makes sense - Thank you.  OK, my workaround was to manually download each one and paste the script via the Arduino IDE. You'll get the same effect if you download directly I'm guessing. Worth noting that one I reckon!!

moodybass, 2015-07-09:
Ahhh makes sense - Thank you.  OK, my workaround was to open each one on here and paste the script via the Arduino IDE. You'll get the same effect if you download directly I'm guessing. Worth noting that one I reckon!!

ajfisher, 2015-07-10:
Yes I noticed this as well a day or so ago but hadn't come up with a way to
resolve it. I will probably build a grunt task to manually copy the files
to the right location from their source or something. This is an issue when
you do an npm install as well.

ajfisher, 2015-07-10:
Reopening this. 

Task: Fix linking of files to the various firmwares so that symlinks are not used. Create a source directory that will manage all the core files and then put them to the right place as part of a build process.

To include AF NP files, ws2812 libs etc.

ajfisher, 2015-07-12:
started a new branch called `flattening` intent is to remove all the symlinks and to use grunt to run a "build" task on the source directory. Will require a reasonable file system restructure so has been moved to a new branch as this will break just about everything.

Might need some testers....

moodybass, 2015-07-12:
@ajfisher - would be more than happy to test & can write up if needed (sorry, I'd help with the code, but a bit of a noob to all this).

ajfisher, 2015-07-13:
Thanks John. Have started the refactor so should hopefully have something
for you to test before I jump on my plane back to AUS tomorrow.

ajfisher, 2015-07-21:
Okay - @moodybass you can test now but I believe this is all working. Have tried it on a couple of laptops and VMs and all appears to be good.

closed 2015-07-21 by ajfisher

moodybass, 2015-07-23:
@ajfisher Had a few attempts on Win7 - worked perfectly each time!  Nice one - thanks Mr Fisher!