ajfisher/node-pixel #37: Intel Edison

kersten, 2015-08-04:
Hello,

I have the mini breakoutboard with the intel edison. Now I wanted to try your library and wired the arduino uno r3 on uart1. I've used your custom firmata and can connect to the arduino, but the strip won't light up. Could you help me with that?

Greetings

ajfisher, 2015-08-04:
Are you using the backpack arrangement?

In this case with the Edison you would have the Edison running node, the
arduino connected to it using i2c.

The arduino then connects to the neopixels.

On the arduino run the backpack custom firmata.

If you look at the setup guide and look at the hardware set up that uses
the raspberry pi, conceptually you're doing the same thing.

kersten, 2015-08-04:
No. I wanted it to try over serial connection. I know this ist not the faster solution, but I got my edison few days ago and tried some stuff.

I check with i2c too, but first I like to get it over serial runnging.

ajfisher, 2015-08-04:
So to confirm, you're going Edison running node talking to arduino over
hardware serial (ie using pins 0&1 for tx/rx)?

If that's the case before you do anything, flash the arduino with standard
firmata and make sure you can talk to it over serial. Use the Johnny five
led-blink example and make sure you can turn pin 13 LED on and off.

Once you've done that, then use the node pixel firmata.

Something to note is that connecting to the arduino like this will not
cause the arduino to reset. As such you may need to wait for node firmata
to find out and make a capability query again. This can take 10-15 seconds
in total so don't break your application too fast, let it sit for a while
to make sure it's connected.

In the mean time, I'll see if I can get an Edison from somewhere to test
with.

kersten, 2015-08-04:
Yes, this is the case.

I already flashed the arduino with the standard firmata and the johnny-five led example works.

what do you mean by waiting? I use the normal johnny-five ready callback. Should I wait another 10 - 15 seconds?

When I'm back home i try to use the led example with your custom firmata, should work too right?

ajfisher, 2015-08-04:
Ok cool. Just making sure all the usual variables are ready. If it's
dropping into the Johnny five ready callback and working then the
electronics is good.

From here, make sure you are using node pixel firmata from
`firmware/build`. Start with the defaults (neopixels attached to pin 6 on
the arduino).

Then do the same thing; LED-blink to make sure I haven't done something
crazy with the custom firmata. Assuming that works, then try the Johnny
five.JS example which should pulse a sequence of pixels along the strip.

kersten, 2015-08-04:
Ok, I will try that. This will be in a couple of hours when I am back home ;)

ajfisher, 2015-08-04:
Is this the setup you're using coupled to a normal arduino?

http://tronixlabs.com/iot/intel-edison/intel-edison-w-mini-breakout-board-australia/

kersten, 2015-08-04:
Yes!

kersten, 2015-08-04:
I got this today, will try that too.

[image]

kersten, 2015-08-04:
Ok, got this running now. I have no idea why, I just started the sketch today again and it was running.

But a strange thing: From time to time the johnny-five lib can't talk to the arduino.

I could test board above because it does not fit on the normal mini breakout board, so I have to get the powering one from sparkfun.

Greetings kersten

ajfisher, 2015-08-05:
So it appears to be working okay now? Would love to see a pic / vid of this
working if possible. I've got an Edison on order but it won't arrive until
next week to test with.

I'm not sure what's going on with the johnny-five -> Arduino side - perhaps
this is something that @rwaldron has seen as I know he's played with the
Edison's a *lot*.

rwaldron, 2015-08-05:
> I could test board above because it does not fit on the normal mini breakout board, so I have to get the powering one from sparkfun.

The SparkFun board shown doesn't connect to the mini breakout, it connects to directly to the Edison, but _does_ require the battery block. Also, if/when you switch to the Sparkfun Arduino Block, you **must** remove Galileo-IO/Edison-IO from your Johnny-Five code and simply specify the port path in use. [Explanation here](https://github.com/rwaldron/galileo-io#sparkfun-edison-arduino-block)

> But a strange thing: From time to time the johnny-five lib can't talk to the arduino

This isn't caused by Johnny-Five, there is likely an issue with the i2c bus initialization—I've experienced the same problem myself. I just reboot the board.

kersten, 2015-08-06:
@ajfisher It is running now, but I have some other problems.

1.
I have multiple strips on pins 6, 9, 10, 11. But only pin6 is working, do I have to update anything?

2.
If try to set the color for all leds to a single color like #FFFFFF the first couple of leds are in random color. 
If I try to set them to a different color like or #FF0000 they will stay random and white (I turned them off and turned the power off, but result is the same). Setting them to #000000 works like expected.

I will do some more tests with the working strip to figure out the color problem.

Is there some function to set the brightness of the leds?

I will post a video when all problems are solved and all leds are running :-)

@rwaldron OK, the console/power block arrives today, but maybe I will not use them right now because my setup is running with normal arduino and the mini breakout board. So I will use them in the next project (garage lightning :-) )

ajfisher, 2015-08-06:
Okay that's great news. Re your other issues:

1. Multi-pin isn't quite supported yet. I've started on the process. You
can join the data out on the last pixel of your strip to the data in on the
next one and chain them that way. Alternatively a PR is always awesome if
you want to tackle this. Otherwise I'm about a week away as won't be
focussing on new dev for a few days (am running a JS IoT workshop this
weekend).

2. That sounds like a power issue. Make sure you're powering the strip with
a clean 5V supply (I'd say NOT the one powering your Edison and Arduino as
well). You can also add a smoothing capacitor across the power and ground
lines to the strip which should help that too.

3. There's not a brightness function as yet - the way I do it is just
modify the hex colour. (As #000000 is off, not "black" - so #880000 is
"half on red" really). Would be cool to be able to modify the brightness
value just using a setting though.

frxnz, 2015-08-06:
> You can also add a smoothing capacitor across the power and ground
lines to the strip which should help that too.

If you're in a static-y environment I would also add a 330ohm resistor on the data line. I've had strips go wonky on me because the first pixel got hit a little too hard.

kersten, 2015-08-07:
@ajfisher Thanks for your answer.

1. Have fun at the workshop :smile:! I might have a look on that later on that multipin stuff, but first I want to get this thing running without any problems.

2. I have an extra external 5V power supply only attached to the strips, nowhere else. I might try the resistor solution.

3. Ok, I understand.

4. What do you think of implementing https://github.com/FastLED/FastLED instead of neopixel? Was there any reason to don't use ist?

I hope I can make a video of that on the weekend.

ajfisher, 2015-08-07:
>    Have fun at the workshop [image: :smile:]! I might have a look on that

Most of the reason is because of all the template use in the firmware makes
it a bit less portable with fast LED. To be honest even the NP lib is
overkill for what's needed as the "hard" processing can be done node side.
I just haven't had time to rip out the rest of the lib while we work on
features etc.

Plan is basically all that's kept on arduino side is a memory buffer that
can just write out as fast as possible and with some very low level
optimisations to shift slabs of memory around / do bulk operations.

kersten, 2015-08-25:
hi @ajfisher,

I've just testet your multipin branch. But I cant get it to work. I have 6 strips attached to a 74HCT245

[image]

This one is connected to the arduino pins of the arduino edison block on pin 11, 10, 9, 6, 5, 3. But But no strip will light up if i start my test blinking program. Could you tell me if this branch is already working?

Thanks

Kersten

kersten, 2015-08-25:
Ah just saw in the readme that there is a limit of 64 pixels for each strip. I have a setup like: 1st strip 20 pixels, 2nd strip 80 pixel, 3rd strip 50 pixels and two are 140 pixels.

Could I get this to work with the current state of the lib?

rwaldron, 2015-08-25:
@kersten can you paste your code here?

kersten, 2015-08-25:
Hi @rwaldron,

here is the working code:

```javascript
var async = require( 'async' ),
	five = require( 'johnny-five' ),
	Edison = require( 'edison-io' ),
	pixel = require( 'node-pixel' ),
	RandomColor = require( 'just.randomcolor' ),
	arduino = new five.Board( {
		port: '/dev/ttyMFD1',
		repl: false
	} ),
	edison = new five.Board( {
		io: new Edison(),
		repl: false
	} ), strips = {
		deck: {
			main: null,
			side: null,
			back: null,
			stairs: null
		},
		house: {
			left: null,
			right: null
		}
	};

function allOn( strip, color ) {
	if ( color === 'random' ) {
		for ( var i = 0; i < strip.stripLength(); i ++ ) {
			var randomColor = new RandomColor();
			strip.pixel( i ).color( '#' + randomColor.toHex().value );
		}
	} else {
		strip.color( color );
	}

	strip.show();
}

function allOff( strip ) {
	//console.log( 'Turn all leds (' + strip.stripLength() + ') off' );

	for ( var i = 0; i < strip.stripLength(); i ++ ) {
		strip.pixel( i ).color( '#000000' );
	}

	strip.show();
}

async.parallel( [
	function ( done ) {
		arduino.on( 'ready', function () {
			async.parallel( [
				function ( done ) {
					strips.house.right = new pixel.Strip( {
						data: 11,
						length: 140,
						board: arduino,
						controller: 'FIRMATA',
						strip_id: 0
					} );

					strips.house.right.on( 'ready', done );
				}, function ( done ) {
					strips.house.left = new pixel.Strip( {
						data: 10,
						length: 140,
						board: arduino,
						controller: 'FIRMATA',
						strip_id: 1
					} );

					strips.house.left.on( 'ready', done );
				}, function ( done ) {
					strips.deck.front = new pixel.Strip( {
						data: 9,
						length: 140,
						board: arduino,
						controller: 'FIRMATA',
						strip_id: 2
					} );

					strips.deck.front.on( 'ready', done );
				}, function ( done ) {
					strips.deck.stairs = new pixel.Strip( {
						data: 6,
						length: 140,
						board: arduino,
						controller: 'FIRMATA',
						strip_id: 3
					} );

					strips.deck.front.on( 'ready', done );
				}, function ( done ) {
					strips.deck.back = new pixel.Strip( {
						data: 5,
						length: 140,
						board: arduino,
						controller: 'FIRMATA',
						strip_id: 4
					} );

					strips.deck.back.on( 'ready', done );
				}, function ( done ) {
					strips.deck.main = new pixel.Strip( {
						data: 3,
						length: 140,
						board: arduino,
						controller: 'FIRMATA',
						strip_id: 5
					} );

					strips.deck.main.on( 'ready', done );
				}
			], function () {
				done();
			} );
		} );
	}, function ( done ) {
		edison.on( 'ready', done );
	}
], function ( err ) {
	if ( err ) {
		console.log( err );
		process.exit();
	}

	console.log( 'app ready' );

	setInterval( function () {
		console.log( 'Off' );

		allOff( strips.house.right );

		setTimeout( function () {
			console.log( 'On' );

			allOn( strips.house.right, '#7036a9' );
		}, 1000 );
	}, 2000 );
} );
```

If I add another ```allOn``` like this

```javascript
setInterval( function () {
	console.log( 'Off' );

	allOff( strips.house.right );
	allOff( strips.deck.back );

	setTimeout( function () {
		console.log( 'On' );

		allOn( strips.house.right, '#7036a9' );
		allOn( strips.deck.back, '#FF0000' );
	}, 1000 );
}, 2000 );
```

it does not work anymore. I have to reset the arduino over the reset button and restart my sketch. Otherwise the arduino will not be reconized anymore by ```arduino.on( 'ready' )``` and I get a timeout error.

rwaldron, 2015-08-25:
```js
var async = require( 'async' ),
    five = require( 'johnny-five' ),
    Edison = require( 'edison-io' ),
...
    arduino = new five.Board( {
        port: '/dev/ttyMFD1',
        repl: false
    } ),
    edison = new five.Board( {
        io: new Edison(),
        repl: false
    } )
```

Remove these: 

```js
 Edison = require( 'edison-io' ),

    edison = new five.Board( {
        io: new Edison(),
        repl: false
    } )
```

This _cannot_ be used with the Sparkfun Arduino Block. With that change, your initialization should look like: 

```js
var async = require( 'async' ),
    five = require( 'johnny-five' ),
    pixel = require( 'node-pixel' ),
    RandomColor = require( 'just.randomcolor' ),
    arduino = new five.Board( {
        port: '/dev/ttyMFD1',
        repl: false
    } ), 
    strips = {
        deck: {
            main: null,
            side: null,
            back: null,
            stairs: null
        },
        house: {
            left: null,
            right: null
        }
    };
```

kersten, 2015-08-25:
ok, tried that no change. Would not make any sense to me removing edison initialisation, I hooked up the GPIO block to it too.

But I saw some other stuff. I only wired one strip now. Init all of them. If I use pin 11 its working fine. If I use pin 10 the strip will light on, but never go off. If I use pin 5 nothing happens and the arduino crashes again.

rwaldron, 2015-08-25:
> Would not make any sense to me removing edison initialisation, I hooked up the GPIO block to it too.

This is not tested and I can't guarantee that it will work :(

I'd like to reduce the case as much as possible to figure out what is going on with the part connected to the neo-pixels.

kersten, 2015-08-25:
OK, I just made this change:

changed this

```javascript
function allOff( strip ) {
    //console.log( 'Turn all leds (' + strip.stripLength() + ') off' );

    for ( var i = 0; i < strip.stripLength(); i ++ ) {
        strip.pixel( i ).color( '#000000' );
    }

    strip.show();
}
```

to this

```javascript
function allOff( strip ) {
	strip.color( '#000000' );
	strip.show();
}
```

Now it works with pin 10 too.

Maybe this could be a timing problem. If I use a for loop and set every pixel its own color, only one strip will work. I just reduced the stripLength to the actual strips that i have to test on my table. But no luck. If I loop through the pixels only one strip works.

https://www.youtube.com/watch?v=LmwKK5lDc_c

ajfisher, 2015-08-26:
So multipin is *really* rough at the moment. I don't think it's ready for
"strange" scenarios like this yet as I can't even guarantee it works
properly on an i2c backpack at this point.

I would focus on using the master branch and trying to get this set up
working with a single strip.

I don't quite understand what this part is doing in this case? Buffering
the output on the pin is going to cause you some challenges for sure
because of the timing aspects of the frames...

ajfisher, 2015-11-25:
Am going to close this off for the moment. Please reopen if needed.

closed 2015-11-25 by ajfisher