ajfisher/node-pixel #40: Adafruit 12 Neopixel ring only half lights up

ismailuddin, 2015-09-04:
If you use the 'Johnny-Five' implementation to light up the LEDs in an Adafruit 12 LED ring, only the second half of the ring lights up. Pixels addressed 0:4, even when instructed to light up, do not actually light up. Any ideas as to why this is occurring?

ajfisher, 2015-09-04:
Can you post your code, circuit diagram and also can you confirm what board
you're using and you have put the custom firmware on the board or backpack?

ismailuddin, 2015-09-05:
I'm using a Hobby Components Arduino Uno board. I've connected the data IN through hole to pin 6 (PWM pin) on the board, the power to +5V on the board; and GND to GND on the board.

https://gist.github.com/ismailuddin/496dc82cabdab1c4ba92

I've confirmed the ring is fully lighting up by running an Arduino script from the Arduino IDE which lights up all the LEDs properly.

I've uploaded the `node-pixel-firmata.ino` script to the board using the Arduino IDE. I've tried operating LED13 pin using this set up and Node REPL, and it works fine like that.

An interesting thing I notice is that if I try to enter the script line by line through the REPL in command prompt (using Windows 10 btw), after I've initialised the JohhnyFive module, I start getting this weird double character effect (so I just run the test.js file from CMD). This might be related to maybe cmder/ConEmu terminal I'm using though, so it may not be relevant.

[image]

ajfisher, 2015-09-05:
Thanks for that. Okay so I'm not sure what your double chars are - looks like local echo on your terminal emulator??

Okay so there's definitely a problem with the code you're posting. I'll give you the explanation first then the code mod.

NodePixel works by decoupling the setting of pixel or strip colours from the display of them. You can think of this like a frame buffer when you're doing any graphics work. This means that you set pixels how you want them which updates the memory locations with the relevant values and then you update the strip using the `show()` command which happens "roughly" all in one go (more on that in a moment).

The next thing you need to consider, and this is more firmata related than anything else, is that if you send lots of messages very rapidly, there's no guarantee on the firmata side that it may get and process all of them as they may get overwritten in the serial buffer or ignored for other reasons (more on that in a sec). This is particularly the case in tight for loops like this. Don't forget that your computer will execute orders of magnitude faster than both the serial baud rate and arduino supports.

Related to all of this is that the `.show()` method is actually blocking on the arduino side because of the way it has to stop any interrupts and then write out to the pin (as neopixels require no interruptions at very tight timings for the pulses that set the values).

All of this is quite a long explanation but basically the problem is your initial show is executing on the arduino and blocking at the same time as you're sending other messages to it to set the colour values. These will get overwritten in the buffer. The reason you see some light up and some not is that the number missing is a measure of how fast your computer is operating compared to the fixed speed of 800ns per pixel it takes to write to it. If you're missing a lot out of the 12 then your computer is really fast, if you did this on a Raspberry Pi then maybe you'd only miss one or two.

I can replicate this on my mac and it skips 7.

Solution - remove the initial `.show()` call and you're all good.

```javascript

pixel = require("node-pixel");
five = require("johnny-five");

var board = new five.Board();

var randomColor = function (low, high) {
	return Math.floor(Math.random() * (high - low) + low);
}

board.on("ready", function() {
	console.log("Board initialised")
    var strip = new pixel.Strip({
        data: 6,
        length: 12,
        board: this,
        controller: "FIRMATA",
    });

    strip.on("ready", function() {
        // do stuff with the strip here.
        console.log("Pixels about to light up")
        // strip.show(); - remove this and it won't block your next pixel writes on the colours.

        for (i = 0; i < strip.stripLength(); i++) {
        	console.log(i);
        	strip.pixel(i).color("teal");
        }
        strip.show(); // this one is fine as you're not writing anything else.
    });

});

```

The rule of thumb I work to is make your `.show()` calls as infrequent as you can get away with and then try not to pass any other messages over to the arduino while it's doing it as it's not guaranteed it will see them. Once `multipin` and `shape` land I'll have a look at what shifting the show method to a promise with an ack or something once completed looks like.

Hope that helps.

ismailuddin, 2015-09-05:
That makes a lot of sense actually! Thanks a lot for clearing that up with me! Good to properly understand how it works! :+1:  :) It all works fine now!

ajfisher, 2015-09-05:
Excellent. Glad it's working. I'll make a note about write conflicts somewhere in the read me too as it's something that comes up occasionally

closed 2015-09-05 by ajfisher