ajfisher/node-pixel #41: Strip show slow when is too much leds

marcoocram, 2015-11-06:
I don't know if is a issue with this library, the original, raspberry or arduino but when i use a lot of leds (>100), when i do a interval animation is showing very slow.

Can i avoid this?

Thanks!

marcoocram, 2015-11-06:
It's when i put a individual led by led color:

for(var i = 0; i < NUM_LEDS; i++) {
        var p = strip.pixel(i);
        p.color(leds[i].color);
    }

    strip.show();

marcoocram, 2015-11-06:
I think that this is the problem: https://github.com/ajfisher/node-pixel/issues/4

ajfisher, 2015-11-06:
Can you post your code please?

Chances are you've run into a similar issue.

marcoocram, 2015-11-06:
It's this one: https://github.com/marcoocram/proyecto_arbol/blob/master/server_caidabola.js

Thanks!

ajfisher, 2015-11-06:
Okay so per #4 I  think I can see where you're issue is - and bear in mind I'm making some assumptions about what your code is doing without having run it.

In this block:

```javascript
                var interval = setInterval(function() {
                    if(active_led - 1 >= 0 && !leds[active_led - 1].active) {
                        leds[active_led].color = "#000000";
                        active_led--;
                        leds[active_led].color = color;

                        colorize();
                    } else {
                        console.log("FIN");
                        leds[active_led].active = 1;
                        clearInterval(interval);
                    }
                }, 100);
            });
```

It looks as though after you get a specific message you're creating this interval that's going to fire every 100ms whilst it counts down through the number of LEDs.

The challenge though is that you call `colorize()` which sends a stream of requests to change the pixel colours and then after that calls `strip.show()`.

So there's a couple of challenges with this:

Sending 150 messages over firmata in a stream is a reasonable amount of data you're trying to push across the wire. Each pixel is 10bytes in terms of message handling etc so 10 x 150 = 1500 bytes = 12000bps.

Do that more than about 5 times per second and you've exhausted your 57600 baud rate. You can up this to give you more headroom but this will definitely always be a ceiling.

So my recommendation is try and only send the messages you *need* rather than all of them as this will remove one of the bottlenecks.

The next thing is the call to `strip.show()`. Again because of the way neopixels work, each pixel in the strip requires approximately 0.1ms to take the colour, latch and propagate the signal. So your _minimum_ refresh time per cycle is about 15ms for 150 pixels so gives you a theoretical maximum of about 60fps. In practice this will be lower because your device is doing other things besides simply flipping bits on a single pin the whole time (like processing that slew of messages coming across the serial connection).

My advice is definitely to only run a `show()` when you have to and to limit the number of messages you have going over the wire. Once multipin drops this will help refresh rate as it will parallelise strips across a whole port and there's definitely designs for sending more efficient data over the wire too.

marcoocram, 2015-11-10:
You're right!

I'm only need modify two leds at time and i was modifing all of the strip, i've change my script and now the animation is possible!!

Thank you!!

closed 2015-11-10 by marcoocram

ajfisher, 2015-11-10:
That's great! Would love to see some pics/vid of what you're doing when you
get a chance. I love seeing how people are using node pixel

marcoocram, 2015-11-10:
This is only a little test Of that im working on: https://youtu.be/9MV4mGS1VMs

The big win challenge: a interactive online chritsmas tree!

ajfisher, 2015-11-10:
That's ace! Love it. Can I share it (are you on twitter)?

marcoocram, 2015-11-10:
of course ;) @marcoocram