ajfisher/node-pixel #42: Implement Interchange support

ajfisher, 2015-11-07:
Per https://github.com/ajfisher/nodebots-interchange/issues/6 implement proper support for Interchange.

adam-back, 2016-01-22:
Hi @ajfisher, we're looking to use interchange to install node-pixel's custom firmata onto Intel Edison's Arduino. The Intel IDE allows us to run Node, but not to flash with a firmata.

Any update on when this could be implemented? Or, are there other command-line tricks you can point me towards so we can install the firmata?

Thanks for your robotics work!

ajfisher, 2016-01-22:
I haven't played with an edison before (still have one sitting in a box unopened in my todo pile).

This might be something easier done in gitter.im/ajfisher/node-pixel and we can probably slap something together. Otherwise, interchange support was on my list for this weekend.

adam-back, 2016-01-22:
http://www.intel.com/content/www/us/en/support/boards-and-kits/000005801.html#4

ajfisher, 2016-09-06:
Now done as part of 0.7.x :shipit:

closed 2016-09-06 by ajfisher