ajfisher/node-pixel #60: Can't get it to work with J5 + Uno

alexcroox, 2016-03-11:
I have a circuit setup that works perfectly when using the Adafruit NeoPixel "strandtest" library uploaded directory to an Uno.

However when I flash that Uno with StandardFirmataPlus in order to control via my rPi the neopixels refuse to do anything. Here is the code:

```
launcher.board = new five.Board({
    repl: false,
    sigint: false
});

launcher.board.on("ready", function() {
    var pixel = require("node-pixel");

    var lights = new pixel.Strip({
        "strips": [{ pin: 6, length: 48 }, ],
        "board": this,
        "controller": "FIRMATA",
    });

    lights.on("ready", function() {

        console.log('Init lights');
        lights.color('#ffffff');
        lights.show();

    });

    lights.on("error", function(err) {
        console.log(err);
    });
});
```

The output is

```
1457696689641 Device(s) /dev/ttyACM0  
1457696689699 Connected /dev/ttyACM0  
Init lights
```

But the pixels don't light up. If I disconnect the Uno from the Pi, back into my laptop and upload the Adafruit example library test again it immediately works. I have to come to conclusion that it's a software issue not a circuit/hardware one.

ajfisher, 2016-03-11:
II'm assuming you're not using a backpack?

You need to either use a backpack OR use the nodepixel firmata. Dive into
this for more deets:
https://github.com/ajfisher/node-pixel/blob/master/docs/installation.md

alexcroox, 2016-03-14:
I intend to use the [MUX Shield 2](http://mayhewlabs.com/products/mux-shield-2) but I took it off while I was having issues with this library and opted to try with the underlying vanilla Ardiuno Uno first.

I'm using StandardFirmataPlus on the Ardiuno and controlling via Raspberry Pi 2 over USB.

alexcroox, 2016-03-14:
OK using the Nodepixel Firmata works well. Sorry for the ignorance around J5/NodePixel Firmatas!

closed 2016-03-14 by alexcroox

ajfisher, 2016-03-14:
No worries at all - glad it's working.

Out of interest - what docs did you go through when you were getting it
working? Reason I ask is this comes up sometimes so a notice about it in
the "pathway" someone follows may help others down the line.

alexcroox, 2016-03-14:
The root of the github repo (README). I actually had this working fine several months ago when I was prototyping all the parts of my new project (and when I'd probably followed the instructions to the letter and actually flashed the custom firmware). However after perma prototyping everything and running through a couple of J5 Firmatas to get all the different sensors working well I assumed since this project mentions J5 a lot that the Firmata from J5 which I was using would work fine. It all comes down to my ignorance and lack of understanding to how Firmata actually works and the core differences between each version of it that these node projects use.

alexcroox, 2016-03-14:
Just found my original path into this repo :) https://github.com/rwaldron/johnny-five/issues/925