ajfisher/node-pixel #63: process out of memory

tosbodes, 2016-05-04:
after I run example at 60 fps, 150 LED length for hour it use a lot of ram and finally stop

$ node rainbow-dynamic.js
1462339794217 Device(s) COM10
1462339794228 Connected COM10
1462339797157 Repl Initialized
>> Board ready, lets add light
Strip ready, let's go
dynamicRainbow

<--- Last few GCs --->

  781117 ms: Scavenge 1399.1 (1458.1) -> 1399.1 (1458.1) MB, 6.5 / 0 ms (+ 4.0 m
s in 1 steps since last GC) [allocation failure] [incremental marking delaying m
ark-sweep].
  782276 ms: Mark-sweep 1399.1 (1458.1) -> 1397.9 (1458.1) MB, 1132.1 / 60 ms (+
 5.0 ms in 2 steps since start of marking, biggest step 4.0 ms) [last resort gc]
.
  783277 ms: Mark-sweep 1397.9 (1458.1) -> 1397.8 (1458.1) MB, 1001.0 / 45 ms [l
ast resort gc].

<--- JS stacktrace --->

==== JS stack trace =========================================

Security context: 000000CDCC3E3AD1 <JS Object>
    1: get [native weak-collection.js:~26] [pc=0000006D3A011E0F] (this=000002450
5A73861 <JS WeakMap>,i=0000008D5932CEF9 <a Pixel with map 000001CD340A3119>)
    2: pixel_color [C:\Users\ADMIN\test\node-pixel\lib\pixel.js:~416] [pc=000000
6D3A016C1D] (this=0000008D5932CEF9 <a Pixel with map 000001CD340A3119>,color=000
0012D17AF3C21 <Number: 1.47456e+007>)
    3: color [C:\Users\ADMIN\test\node-p...

FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - process out of memory

ajfisher, 2016-05-05:
Interesting. Don't suppose you can run that on the standard strip example
that just shifts the pixels along?

ajfisher, 2016-05-08:
OK. So I ran a strip using an Uno and node pixel firmata with about 45
pixels on it at 40fps for approximately 8 hours yesterday without any
trouble.

Will have access to a longer strip tonight so will have a go with that and
see if I can get it to fail.

ajfisher, 2016-09-06:
Closing this off due to age and lots of testing since then not able to replicate the issue.

closed 2016-09-06 by ajfisher