ajfisher/node-pixel #66: StandardFirmataEthernet vs node_pixel_firmata

alfredoborrero, 2016-06-27:
Hi, currently my Home Automation is running 5 Arduino Boards connected to my Ethernet network (with Ethernet shields) and I use Johny Five to control all the devices (door sensor, motion sensor, relays, flowmeter, temperature, etc). 

Now I want to add Addressable RGB LED Strip to my house and realized that you guys are providing a new "firmware/sketch" to upload to the Arduino board. If I upload this new Firmata i will loose Ethernet support, do you guys have any Firmata that provides Ethernet support?

I noticed that you guys added couple of lines to he StandardFirmata

Like:

```
//#include "./Adafruit_NeoPixel.h"
#include "lw_ws2812.h"
#include "ws2812.h"

    case PIXEL_COMMAND:
        if (argc > 0) {
            // maybe bounce the first command off here.
            process_command(argc, argv);
        }
        break;

```

Do you think if i add these new lines my system would work? I then i would be able to control my lights?

ajfisher, 2016-06-27:
That would probably work though I'm not sure if you'll run into any other
issues. My recommendation in this case would be to get something like an
arduino nano or pro mini and run node-pixel off that as an I2C backpack and
then connect the backpack to whatever you're using running network firmata.

ajfisher, 2016-09-12:
Closing this out.

closed 2016-09-12 by ajfisher