ajfisher/node-pixel #68: I2C backpack wiring

taktran, 2016-07-13:
Hi, I'm following the [Hardware installation](https://github.com/ajfisher/node-pixel/blob/master/docs/installation.md#hardware-installation-1) for the I2C backpack, but I I'm a bit stuck. I may have missed something really obvious =/

I've wired it like the [diagram](https://github.com/ajfisher/node-pixel/blob/master/docs/breadboard/i2c_backpack_bb.png), but with a raspberry pi 2 model b, and an arduino pro mini. I've also flashed the pro mini with the backpack arduino sketch.

1st question, is what pin is the data pin from the LED strip (`Din` on the diagram) supposed to connect to? The diagram connects to `RX0`, but the docs says to `data line to an arduino digital pin`, which I presume to be pin 6, seeing as it also says `The default configuration assumes PIN 6 (NOTE this is NOT configurable)`.

I tried both, and nothing seems to happen. I ran the following simple script on the raspberry pi:

```
var pixel = require('node-pixel');
var five = require('johnny-five');
var board = new five.Board();

board.on('ready', function () {
  var strip = new pixel.Strip({
    pin: 6,
    length: 10,
    board: this,
    controller: 'I2CBACKPACK'
  });

  board.repl.inject({
    strip: strip,
    showColor: showColor
  });

  strip.on('ready', function () {
    console.log('Pixels ready\n');
    const p = strip.pixel(0);     // get second LED
    p.color("#0000FF");

    const p2 = strip.pixel(191);     // get second LED
    p2.color("#FF00FF");
  });
});
```

But all I get is this output, with the script stalling:

    1468448989073 Board Looking for connected device

The nano, has worked with the much simpler firmata sketch.

Any tips, would be amazing =)

ajfisher, 2016-07-14:
Okay so the pertinent pit is from the readme:

> On a backpack, the strips are defined sequentially from pin 0-7 on the backpack.

On a backpack you can't specify which pins you use, only the lengths of them. 

So if you change:

```javascript
  var strip = new pixel.Strip({
    pin: 6,
    length: 10,
    board: this,
    controller: 'I2CBACKPACK'
  });
```

To:

```javascript
  var strip = new pixel.Strip({
    strips: [10],
    board: this,
    controller: 'I2CBACKPACK'
  });
```

And plug the data line into your strip on Pin 0 then you should be good to go.

By way of background, the reason this is like this is because to try and get the highest possible speed, we're switching on a whole port (Port D in this case) which is pin 0-7 - this gives some efficiency when you're pumping data out as fast as what we need to to meet the timing constraints of the WS2812 leds (and get reasonable lengths of LEDs).

Check out this for a direct example and it uses an RPi as well.

https://github.com/ajfisher/node-pixel/blob/master/examples/johnnyfive-i2c.js

taktran, 2016-07-14:
Thanks @ajfisher. Unfortunately, I still couldn't get it working 😢 

Here's [some photos of it wired up](https://goo.gl/photos/1R97JeZK617DTQr46) in case you see something I'm obviously doing wrong. It's also got the [pimoroni explorer hat pro on top](https://shop.pimoroni.com/products/explorer-hat) in case you are wondering.

I connected:

* nano `A4` -> rpi `SDA`
* nano `A5` -> rpi `SCL`
* nano `RX1` -> LED `Din` (I also tried connecting to nano `TX0`
* nano `GND` -> rpi `GND` -> LED `GND` -> 5V mains `-ve`
* rpi `5V` -> nano `VCC` -> LED `5V`-> 5V mains `+ve`

Basically it's the same issue as before, as when I run the [script](https://github.com/taktran/light-strip/blob/master/src/lights-backpack.js) on the rpi, it stalls:

```
$ node src/lights-backpack.js
1468536711074 Board Looking for connected device
```

I tried wiring it up to an arduino uno too, but that had the same outcome. Not really sure how the `node-pixel` actually looks for the board, and what it is waiting for.

Next up, I'll try:

* Without the explorer hat off
* Just using firmata on the rpi
* Anything else you suggest?

ajfisher, 2016-07-14:
Ahh okay I know what's going on.

So you need to go one of two paths.

The backpack is talking I2C so you can run raspi-io and talk to it directly
OR you need to connect the RPi to an arduino and then have that connect to
the backpack over I2C.

Whilst the backpack is a nano, don't think of it as one - it's a chip
that's enabling I2C in this case.

That's why you're seeing the "Looking for connected device" part.

Try using raspi-io eg

```
var raspi = require("raspi-io");

var board = new five.Board({
  io: new raspi(),
});
```

And go from there.

taktran, 2016-07-19:
Success! 🎉 

[image]

Thanks a bunch @ajfisher!

I've created a repo with the code I used to test it: https://github.com/taktran/raspi-lights

I was initially stuck on my [previous repo](https://github.com/taktran/light-strip), but I think there was too much cruft in there for developing, which meant npm kept failing to install from the wifi cutting off.

Thanks again, can't wait to get started! =)

closed 2016-07-19 by taktran

ajfisher, 2016-07-19:
Awesome stuff @taktran - can't wait to see what you build with it - that's a nice number of LEDs you've got there on that roll :)