ajfisher/node-pixel #70: Is 500 pixels really the limit for the I2CBACKPACK?

taktran, 2016-07-20:
Carrying on from https://github.com/ajfisher/node-pixel/issues/68, I tried setting the number of pixels to 300, but it seems to cut off at the 256th pixel ie,

```
  // Set up strip
  strip = new pixel.Strip({
    color_order: pixel.COLOR_ORDER.GRB,
    board: this,
    controller: "I2CBACKPACK",
    strips: [300]
  });

  strip.on("ready", function() {
    // Last index that works
    var index = 255;
    var p = strip.pixel(index);
    p.color("blue");

    // 1st index that doesn't work
    var index2 = 256;
    var p2 = strip.pixel(index2);
    p2.color("red");

    strip.show();
  });
```

In the readme and the [code](https://github.com/ajfisher/node-pixel/blob/master/lib/pixel.js#L187), it seems like the maximum should be 500. Has this ever been tested? How is this calculated?

----

Edit: Added set up code

ajfisher, 2016-07-20:
Interesting. Reviewing the code(admittedly on my phone) it doesn't (and
shouldn't be) get set on the firmware. The only restriction on there is it
can allocate the memory...

Will dig further.

ajfisher, 2016-07-21:
I've had a bit more of a look at this and will need to dig a lot further as
I'm not sure what's going on. There's nothing that looks like an issue on
the surface of this. As such I need to get to a large number of neopixels
and my debug stuff which I can't do until Sunday (after NodeBots Day here
in AUS).

Hopefully that's ok?

taktran, 2016-07-21:
Of course, that's fine 😄  Thanks a bunch for looking into it!

Have fun at NodeBots Day 🎉

ajfisher, 2016-07-29:
Okay - So I've been having a play with this more this week and I can't replicate this particular problem. Specifically I've used a strip of 360 LEDs and can get them all to light up without any issue at all.

Something to try is this:

```
 strip = new pixel.Strip({
    color_order: pixel.COLOR_ORDER.GRB,
    board: this,
    controller: "I2CBACKPACK",
    strips: [300]
  });

  strip.on("ready", function() {
    // Last index that works
    strip.color("blue");
    strip.show();
  });
```

And see if that works first of all. It might be that you have a dead pixel in your strip which is stopping the propagation of the data through the strip... I know that's happened to me a few times (quite a few on most recent project)

Here's mine working with current release node-pixel etc: https://twitter.com/ajfisher/status/758258493094846464

taktran, 2016-07-31:
Thanks for looking into it @ajfisher.

Unfortunately, that code has the same problem. I don't think it's a dead pixel though, as I've gotten all the pixels to work using an arduino (not with i2c though, using [a modified version of this script](https://github.com/taktran/light-strip/blob/master/arduino/simple/simple.ino)).

raspi setup:

[image]

arduino setup:

[image]

Are you running the i2c backpack from a raspi or a laptop? I couldn't figure out how to get it running on my mac.

ajfisher, 2016-07-31:
Hmm - can you post your config file in /boot on the raspi? I wonder if it's
something to do with the I2C speed settings.

I have run this on both desktop with arduino + I2C Backpack and then
Raspi + I2C backpack and all is good so I wonder what it can be.

As we've discounted the strip being an issues let's try a couple of other
options.

   - Get 2 arduinos and run one with firmata talking to your Mac and the
   other running the I2C backpack code and try that.
   - Do the same thing from your Raspi but use USB to talk to the arduino
   running firmata and make sure that works.
   - Post your config file and we'll see if there's an I2C issue

Also what version of Raspbian are you running and what type of RPi are you
using?

taktran, 2016-08-01:
Ok, this is my boot config

```
$ cat /boot/config.txt
# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

dtparam=i2c_arm=on
dtparam=i2c=on
dtparam=i2c_arm_baudrate=100000
```

Working on rpi 3, and running on jessie

```
$ uname -a
Linux tpi 4.1.18-v7+ #846 SMP Thu Feb 25 14:22:53 GMT 2016 armv7l GNU/Linux
```

Will try out the other set ups and let you know.

ajfisher, 2016-09-06:
Going to close this off as can't be replicated but jump into gitter.im/ajfisher/node-pixel and we can debug further live.

closed 2016-09-06 by ajfisher