ajfisher/node-pixel #71: Error: EIO: i/o error, write

nkolban, 2016-08-28:
Running a backpack on an Arduino Nano, all works fine for a period of time and then eventually an exception is thrown as follows:

```
fs.js:658
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EIO: i/o error, write
    at Error (native)
    at Object.fs.writeSync (fs.js:658:20)
    at Bus.i2cWriteSync (/mnt/cloud/projects/j5/node_modules/raspi-io/node_modules/raspi-i2c/node_modules/i2c-bus/i2c-bus.js:340:13)
    at I2C.writeSync (/mnt/cloud/projects/j5/node_modules/raspi-io/node_modules/raspi-i2c/lib/index.js:332:34)
    at Raspi.i2cWrite (/mnt/cloud/projects/j5/node_modules/raspi-io/lib/index.js:559:19)
    at Strip.Controllers.I2CBACKPACK.strip_color.value (/mnt/cloud/projects/j5/node_modules/node-pixel/lib/pixel.js:295:26)
    at Strip.colour.Strip.color (/mnt/cloud/projects/j5/node_modules/node-pixel/lib/pixel.js:375:14)
    at null._repeat (/mnt/cloud/projects/j5/test.js:22:9)
    at wrapper [as _onTimeout] (timers.js:275:11)
    at Timer.listOnTimeout (timers.js:92:15)

```
Restarting the app, all continues.  It seems that this is a transient error.  Can the module handle such errors gracefully?  Perhaps simply retry some number of times?

My environment is:

* Raspberry Pi 3
* Node: v4.5.0
* Backpack: Arduino Nano

ajfisher, 2016-08-29:
Yep so I've seen this issue as well recently. What it appears to be is the low level RPi I2C controller throwing a write error when it hits a sync issue.

There's a stack of things that can cause this and I've been trying to catalogue where all the problems can come from - the most common appears to be this and only sort of happens under high throughput scenarios (IE you're sending a LOT of pixel data very rapidly at at least moderate refresh rates).

- A write occurs to send a pixel value
- For some reason (probably OS related) that write holds for a moment (eg because of GC, the RPi just "doing" something else within the OS etc)
- You send another write command but the first hasn't completed.
- You then get a SYNC error on the I2C bus

I believe this is because the I2C bus is waiting for a response and you make another request before it has closed.

I've been trying to think of graceful ways to handle this and I'm tempted to create an "i2c_error" event that can get emitted if this gets detected and then you can decide what you want to do in this case.

I've had plenty of scenarios with all the Pixel work I do where sometimes I don't care if one write fails, I just drop it and move on (like a UDP style data stream) because dropping a pixel in a 500pixel frame doesn't matter as it will get caught on the next one. But other times it's a status indicator so you **really do** want to know it's failed.

I think creates a reasonable balance between the pair of them, it catches the error so your application won't bail and then responsibly emits an event that can be used to do additional handling if you need it.

How would this work in your case @nkolban 

Also looking for feedback on this as an approach - pin @frxnz ??

nkolban, 2016-08-29:
Are the write requests to the backpack through I2C serialized?  Can we not block on a write request until it has completed so we don't attempt to perform two write requests simultaneously?

ajfisher, 2016-08-29:
From what I understood of what raspi-io and wiring pi were doing, I
actually thought that was the case already.

Regardless though, a single write failure shouldn't terminate your whole
application unless it's truly a critical write. Which would be a hell of a
situation for a Neo pixel! ;)

So that's what led me to this solution, which is to be a bit more defensive
then try to handle it a bit better at a lower level (which will take a lot
more poking about to find a solution).

ajfisher, 2016-08-29:
Hmm trying to find the code where I came up against this directly most recently... might actually be **on** a robot somewhere...

Regardless here's my thinking.

Make all calls like this into a try statement:

```
pixel.io.i2cWrite(pixel.i2c_address, data);
``` 

becomes:

```
try {
  pixel.io.i2cWrite(pixel.i2c_address, data);
} catch (e) {
  if (e instance of Error && e.name == "EIO") {
    this.emit("i2c_write_error");
  } else {
    throw e;
  }
}
```

This would also happen nr lines 280, 294 & 270.

So specifically we catch this EIO write sync error and if it's something else then we still error. Under this scenario we catch it and emit a message instead and let the user of the object decide what they want to do (eg try again, fail etc).

I'd need to modify some bits in the `Pixel` class to deal with that as it's not an event emitter but that's fine I can route it to come off the `Strip` in some form anyway.

frxnz, 2016-08-29:
Sounds like a reasonable solution to me. I've also run up against this on an RPi.

I suppose in an animation where you're plugging away at 60fps you're always going to ditch the frame if an error throws, so timing of the event isn't really critical. For other situations it might make sense to emit the `i2c_write_error` with the data you're trying to write. That way you can retry even if you don't know exactly where you were in the application.,

ajfisher, 2016-09-13:
Okay change I just pushed should implement this - can peeps with an RPi give this a go?? On error it emits "np_i2c_write_error"

ajfisher, 2016-09-15:
Looks like this set of mods appears to be working okay... going to publish with this as changes in 0.8.1

justbill2020, 2017-01-25:
i'm constantly getting this error... i have the SDA pin connected on pi zero to the SDA pin on an uno r3 the SCL is to SCL respectively as well. 

the data pin for the strip is connected to D0 

here is my code
```javascript
var five = require("johnny-five");
var pixel = require("node-pixel");
var raspi = require("raspi-io");

var opts = {};
opts.port = process.argv[2] || "";

var board = new five.Board({
    io: new raspi()
});
var strip = null;

var fps = 20; // how many frames per second do you want to try?

board.on("ready", function() {

    console.log("Board ready, lets add light");

    strip = new pixel.Strip({
        color_order: pixel.COLOR_ORDER.GRB,
        board: this,
        controller: "I2CBACKPACK",
        strips: [5],
    });

    strip.on("ready", function() {

        console.log("Strip ready, let's go");

        var colors = ["red", "green", "blue", "yellow", "cyan", "magenta", "white"];
        var current_colors = [0,1,2,3,4];
        var current_pos = [0,1,2,3,4];
        var blinker = setInterval(function() {
                strip.color("#000"); // blanks it out

                for (var i=0; i< current_pos.length; i++) {
                    if (++current_pos[i] >= strip.length) {
                        current_pos[i] = 0;
                        if (++current_colors[i] >= colors.length) current_colors[i] = 0;
                    }
                    strip.pixel(current_pos[i]).color(colors[current_colors[i]]);
                }

                strip.show();
        }, 1000/fps);

    });
});
```

```
1485339453349 Device(s) RaspberryPi-IO  
1485339453548 Connected RaspberryPi-IO  
1485339453619 Repl Initialized  
>> Board ready, lets add light
INFO: Default gamma behaviour is changing
0.9 - gamma=1.0 - consistent with pre-gamma values
0.10 - gamma=2.8 - default fix for WS2812 LEDs
Strip ready, let's go
fs.js:786
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EIO: i/o error, write
    at Error (native)
    at Object.fs.writeSync (fs.js:786:20)
    at Bus.i2cWriteSync (/home/pi/smart-strip/node_modules/i2c-bus/i2c-bus.js:348:13)
    at I2C.writeSync (/home/pi/smart-strip/node_modules/raspi-i2c/dist/index.js:297:37)
    at RaspiIOCore.i2cWrite (/home/pi/smart-strip/node_modules/raspi-io-core/dist/index.js:707:19)
    at Pixel.value (/home/pi/smart-strip/node_modules/node-pixel/lib/pixel.js:677:26)
    at Pixel.colour.Pixel.color (/home/pi/smart-strip/node_modules/node-pixel/lib/pixel.js:766:18)
    at Timeout._onTimeout (/home/pi/smart-strip/example.js:42:45)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
```

also as an FYI, I've loaded the strandtest.ino sketch directly to the uno and it works. its not sure how to even begin troubleshooting at this point it seems there was a fix for that but this happens non stop it never lights up anything. i've attempted adjusting the 

my environment is:
RPi Zero with 1/11/2017 build of Raspbian LITE
nodejs: v6.9.1
npm: v4.1.1
backpack: Uno R3

ajfisher, 2017-01-25:
This looks like it could be something in raspi-io or even further back in
the I2C library. I haven't got a pi zero able to test this side but will
update to latest raspbian image today and give this a go.

Normally the write sync issue is to do with 2 messages hitting the write
buffer at the same time and colliding.

What you could try is writing an on error handler, testing if the err is an
instance of EIO then just passing on it. This should allow the code to
continue execution and you can see what's going on a little more and might
help debug it.

justbill2020, 2017-01-25:
I did a try catch and it just loops the error and nothing else happens

ajfisher, 2017-01-25:
Ok, that sounds even more like an I2C issue. You definitely flashed the
backpack firmware to the Uno as well?

justbill2020, 2017-01-26:
Yes but I do feel like I must have missed something... I flashed the backpack.ino sketch to the uno using the Arduino app... I've gone through the setup multiple times from scratch based on the readme on your repo. Even changed the baud rate

-

ajfisher, 2017-01-26:
You shouldn't need to make any baud rate changes as that's for the serial
version (firmata version)

ajfisher, 2017-01-26:
Okay I've just tested this and can't see any issues. Here's my set up:

## UNO:

Flash the Uno with the node-pixel backpack firmware on your laptop.

```
interchange install node-pixel -a uno
```

Make sure there's no errors from that. Then wire it up per diagram here:

[image]

Ensure that you have SDA and SCL the right way around. Also ensure that even if you are powering the Uno from somewhere else (advisable), you MUST have grounds connected between your Arduino and your RPi.

Also note that it is possible you are seeing some noise and as such you want to connect a 4.7K Ohm pullup resistor on both the SDA and SCL lines (just put between 5V and each of those lines - eg

[image]

## RPi set up

Here's my set up I've tested this with today.

Raspbian from Nov 2016
RPi 3 (couldn't find an earlier one to try and don't have a Zero I can use to test)

Node 6.2.4

### Install dependencies

```
npm install johnny-five node-pixel raspi-io
```

### Relevant config

Boot config /boot/config.txt

```
dtparam=i2c_arm=on
dtparam=i2c=on
dtparam=i2c_arm_baudrate=100000
```

Ensure I2C is enabled via `raspi-config`

Make sure you can "see" the Backpack using:

```
i2cdetect -y 1
```

Output should look like:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- 42 -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Which indicates that there is an I2C device on address `0x42` (which is expected as that's the default address a node-pixel backpack uses).

### Example code:

```
var five = require("johnny-five");
var pixel = require("node-pixel");
var raspi = require('raspi-io')

var board = new five.Board({
    io: new raspi(),
});

var strip = null;

var fps = 20; // how many frames per second do you want to try?

board.on("ready", function() {

    console.log("Board ready, lets add light");

    strip = new pixel.Strip({
        color_order: pixel.COLOR_ORDER.GRB,
        board: this,
        controller: "I2CBACKPACK",
        strips: [8],
    });

    strip.on("ready", function() {

        console.log("Strip ready, let's go");

        var colors = ["red", "green", "blue", "yellow", "cyan", "magenta", "white"];
        var current_colors = [0,1,2,3,4];
        var current_pos = [0,1,2,3,4];
        var blinker = setInterval(function() {

            strip.color("#000"); // blanks it out

            for (var i=0; i< current_pos.length; i++) {
                if (++current_pos[i] >= strip.length) {
                    current_pos[i] = 0;
                    if (++current_colors[i] >= colors.length) current_colors[i] = 0;
                }
                strip.pixel(current_pos[i]).color(colors[current_colors[i]]);
            }

            strip.show();
        }, 1000/fps);
    });
});
```

Then run:

```
sudo /path/to/node example.js
```

And my backpack works as expected.

My feeling is that the issue may be related to one of the following:

* The I2C device isn't available (i2c detect should show you if it is or isn't)
* There's noise on the line - try putting in 4.7K pullup resistors across SDA and SCL - if you have multiple I2C devices this is absolutely mandatory
* There's an issue with the firmware - make sure you flash using interchange unless you run the Grunt build scripts to get the proper firmware to use in arduino
* You've got some sort of permissions issue

One final thing you can try is if you have 2 arduinos, wire them up together per the example here: https://github.com/ajfisher/node-pixel/blob/master/docs/johnnyfive-i2c.md and see if you can get the backpack working. This should give you a good indicator if the backpack is working properly and that would indicate a wiring or RPi issue.

** Reopening Issue for tracking **

justbill2020, 2017-01-26:
I will try all this and report back... I did not use interchange and will add the pull up resister

ajfisher, 2017-01-26:
Try interchange as a first step as the build process for the firmware is a
little fiddly and you may not have got everything if you try to do it
manually.

justbill2020, 2017-01-27:
i wanted to update you on where i currently stand... i had to give up on the pi zero which is a major let down considering in my excitement and anticipation of doing this project i got 3 pi zeros for $5 and they are shitty paperweights since i can't use them for this. I was able to use interchange (although it wasn't working with the interchange command so i had to run `node cli.js install node-pixel - uno` it flashed and on the pi 2b it worked... now i just have to figure out the best way to deal with voltage drop... i'm probably going to have to run 12v to 12 to 5 v hacked adapters to drop to 5v every meter... this sucks cause i'm trying to use this as under cabinet lighting... but that's a different issue for a different day.

ajfisher, 2017-01-27:
Yeah I have only had a PiZero in my hands for a very short time and was somewhat underwhelmed which is disappointing. 

Regarding interchange - did you install with -G option to install globally (alternatively put `./node_modules/.bin` on your path)

Regarding your voltage drop - if you have a 5V 3A supply you should be able to go approx 50 LEDs without needing another supply - what's your density look like?

Just on that, I actually typically run my LEDs on 3.3V because the WS2812s are plenty bright even at that voltage and rarely run them at "full tilt" - this tends to give me **much** bigger runs that I can accomplish (eg a massive Dome Bot that has 500 pixels on it from one power supply)...

justbill2020, 2017-01-27:
so are you pulling 3.3v from the arduino or are you dropping the voltage of 5v i have a 5v 15a power supply and i'm using this LED strip. http://a.co/aNUd8Hf

ajfisher, 2017-01-30:
Small runs (up to about 30 or so) I just run right off the Arduino 3.3V VOUT (which is a bit dodgy but so long as you're not going full white on all LEDs you can get away with it). For anything more than that or for situations where you know you'll be going white a lot then I use a dedicated power supply. I have some reasonably sized 3.7V ones which I use but 3 or 5V both work.

So with your white turning to red problem you're probably just getting a bit of sag through the run of LEDs. Something you can try (especially with a nice beefy 15A supply!) is to simply "inject" some power again along the strip.

What I do is solder a power and ground line onto the strip directly (you don't need to cut it, just solder it onto the pads on the strip) every 50 LEDs and then bring that back to your power supply. That way you're providing a more "steady" supply in parallel to the strips of LEDs because each LED will introduce just a little bit of drop on the power line as it flows through. 

To test this you can just wire both end of your strip back to power and ground then light it up and see you should get nice bright ends and it sag in the middle. Just add more points as needed :)

theboxer, 2018-02-10:
I'm having an issue with this error as well. I have rPI 0W with arduino nano as a I2C backpack. My led strip has 300 leds and when I try to set them all through for loop (I know I can use strip.color to set color for all leds, but I'm trying to set different color to each led), I got this error several times...

Any suggestion, how I can fix it?

lePioo, 2018-07-20:
Same here with a PI3B and Arduino Uno i2c backpack with 210 leds on a strip.
First I got an error reported by nodebots-interchange when trying to flash, it was related to an update of node-serialport and its new way of using parsers. Once that resolved I could flash the board with interchange install git+https://github.com/ajfisher/node-pixel -a nano but I always ended with [issue 92](https://github.com/ajfisher/node-pixel/issues/92). Even if the configuration failed I was able to detect my arduino with i2c detect and was able to use the api. But with high rate speed (20ms) and my 210 leds it crashes the way this issue describes in less than a minute.
I'll try to perform clean os install on my rpi soon, retry the whole thing and let you know

ajfisher, 2018-07-20:
So this isn't a node-pixel error, it's an error in the write buffer in the
RPi - basically this can occur if you're pushing a lot of data and
infrequently you'll get a write error. Best thing to do is just catch the
EIO type error and then handle it (usually by dropping it but for your
application you may have a need for finessing that).

ajfisher, 2018-07-25:
Going to close this out as the fix is to catch the `EIO` error and either handle it or drop it as needed.

commit 051014df (2016-09-15): 0.8.1 add .off capability fix bugs with IO writes, fixes #71, closes #61
closed 2018-07-25 by ajfisher