ajfisher/node-pixel #72: Ability to reset the Backpack or recover from interruption ...

nkolban, 2016-09-02:
I'm using the project to drive neopixels using an Arduino backpack driven by CHIP.  When all works, it works well.  My front end is a Node.js/Johnny-Five app and, during testing, I find that when I interrupt the app, there is a "probability" that when I restart the app the Arduino is "confused".  My guess is that the Arduino logic (I haven't looked yet) is some kind of state machine and depending where we interrupt the messages sent via I2C to it, it gets itself into a "confused" state.  I'm wondering if there isn't something we can do about this?  Perhaps some kind of reset function or else a "state timeout" that returns it to a freshly booted state if there has been confusion?

I'm thinking that the issue lives in the Arduino code as if I press the physical reset button on the Arduino, and try my app again, it starts to work once more.

nkolban, 2016-09-02:
Here are some guesses as I start to look through the Arduino logic in "backpack.ino" ... I'll update this post as thoughts arrive.

Looking at line 46 ... we issue a Wire.read() which is a blocking call with timeout.  The default timeout is 1000 msecs.  I notice we don't check the response from read() for "-1" which would indicate a read error.  We might want to check that value and if we get a "-1" abandon the request.

ajfisher, 2016-09-04:
@nkolban interesting - do you want to give that a crack and see how it rolls against your setup. I just got notification my CHIP is still a long way from being shipped so would be good to test this idea against your setup.

ajfisher, 2016-09-13:
@nkolban I can't test for this for at the moment but the code you'd need is this:

@47
+++
```
if (received_bytes[i] == -1) {
    return;
}
```

Should be it. If you can drop that in, compile and give it a crack that would be amazing.

nkolban, 2016-09-13:
Made a note to test ... will probably be weekend now ... but its on my list.