ajfisher/node-pixel #73: Add support for SK6812

ajfisher, 2016-09-05:
As these are starting to become more available and are largely the same as the WS2812, add direct support for these by being able to shift a little faster on the HIs.

ebrearley, 2017-05-28:
+1I have 300 of these in my living room, installed them yesterday. I'd love to help out by adding this support, but on account of being a complete newbie and only understanding 20% of things right now, I'll just leave my plus one here 😝

cotejp, 2017-11-12:
Is there really a difference? [Some people say](https://hackaday.com/2017/01/25/ws2812b-led-clones-work-better-than-originals/) that SK6812 are simply clones of WS2812b which would make them functionally equivalent.

I believe I'm currently using SK6812 LEDs from Adafruit and they work fine with the library.

TigerChi, 2018-04-22:
+ 1 
Now I use Johnny-five control most of device, and another Arduino board to control SK6812 by UART.... 
If node-pixel can support RGBW it will be great !