ajfisher/node-pixel #79: Firmata for MKR1000: fatal error: avr/io.h: No such file or directory  #include <avr/io.h>

StephenShh, 2016-09-12:
Hey, AJ. I seem to keep getting the following error in Arduino IDE any time I attempt to flash the Firmata to a MKR1000. What might I be missing. Thanks in advance.

[image]

ajfisher, 2016-09-12:
That's because the mkr1000 isn't an avr board. Basically this chipset is
using the ARM Cortex chipset so it can't find that reference

Not sure exactly the path to fixing that as that library is dealing with
all the cpu and memory definitions which we'd need equivalent mapping for
on the mkr1000.

I'm sure it exists. Perhaps a quick search for equivalent for that target
and then change the reference in lw_ws2812.h and see if it builds? If it's
as easy as that we can put a pre processor directive in... :fingers crossed:

ajfisher, 2016-09-12:
Okay so you can try to replace that include with `<sam.h>` - I'm not sure I like your chances though because of all of this:

```
#include <avr/interrupt.h>
#include <avr/io.h>
#ifndef F_CPU
#define  F_CPU 16000000UL
#endif
#include <util/delay.h>
```

Basically the interrupt header deals with how we suspend the interrupts (that's used to suspend interrupts when you shift the frame to the pixels) and then the io.h is used for all the pin masking - that stuff *probably* does exist in the sam.h file as it's fairly common.

However util/delay.h holds all the stuff pertinent to the timer loops which is then written in avr specific assembly to get the timing on the pixels absolutely spot on.

My recommendation is this:

Unless you feel a burning need to do a port from avr to sam for all of this (I can actually see upside to this in terms of learning how the internals of the chip works!) I'd probably hand off all of this to a "backpack" version of node-pixel. Simply get a pro-mini or nano which will cost you a couple of bucks, put node-pixel backpack on it and then talk to the the backpack from the mkr1000 using I2C. You will be able to control a LOT of pixels and you will be up and running *really* fast.

ajfisher, 2017-01-19:
Closing this out as there's no impetus for doing a port from AVR to SAM for this and there's a more direct approach which is to use a backpack option.

closed 2017-01-19 by ajfisher