ajfisher/node-pixel #92: Can't Configure Device. Arduino Uno Backpack

hironroy, 2017-01-24:
Hi. Thanks for the great work. I'm trying to get the i2c firmware onto an Arduino Uno. Unfortunately, I'm encountering `Can't configure device. Did you remember to set your backpack into config mode?` after `flash complete` when running `interchange install node-pixel -a uno`

 I've tried lifting D2 w/ 5v, but it doesn't change the behavior. By installing hc-sr04 firmware, I was able to confirm that lifting D2 does in fact place the Uno in config mode. Any thoughts as to what might be going wrong?

ajfisher, 2017-01-24:
@hironroy I know exactly what is happening here - I had to remove the configuration code from the NodePixel firmware as part of a big refactor I did a couple of months ago. As such the current firmware can't get configured by interchange. Having said that the only configuration that is relevant for node-pixel backpack is the setting of the I2C address. 

This is why I haven't re-integrated it as I feel like there might be another way to do this without needing all of the interchange library (which means less memory available for LEDs!) though I'm yet to work all of that through.

In the short term:

- The I2C address is `0x42` by default.
- You can change it in `firmware/src/controller_src/backpack/backpack.ino` at the top if you want to but you'll need to rebuild the firmware manually (just run `grunt build` and then open up the firmware in arduino and flash manually).

Am going to open another ticket now that reintroduces this behaviour so close this off if this is sufficient answer for the moment.

ajfisher, 2017-01-24:
New tracking issue raised at #93

hironroy, 2017-01-24:
Thanks for the speedy reply! So, I went ahead and built via grunt. And then uploaded backpack.ino (w/ the default 0x42 i2c address) onto the UNO. I enabled i2c on my Raspberry PI, but when I run `sudo i2cdetect -y 1` on the Pi, no devices are discovered. I've attached SDA and SCL from the PI to A4 and A5 respectively. Any thoughts?

ajfisher, 2017-01-24:
I have noticed that every once in a while i2cdetect may take a bit of time
to discover the device. What I've done previously is boot it up with the
backpack attached and then just check the following:

* Make sure these lines exist in your boot config

```
dtparam=i2c1=on
dtparam=i2c_arm=on
```

* Make sure the I2C module is loaded (both i2c_bcm... and i2c_dev): `lsmod
| grep "i2c_*"
* run `sudo i2cdetect -y 1` and see if the device is there.

And it should work...

hironroy, 2017-01-25:
Unforunately, after I do that, I still end up with:

```
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```

ajfisher, 2017-01-25:
Hmm. Try reflashing the board again. Outside of that - if you can try
connecting to it from a different board - eg use another arduino to talk to
it over I2C and make sure that's working it might help validate what is
going on.

I'll have a crack at this a bit later today (but am a bit distant from an
RPi) to test it out my side as well.

hironroy, 2017-01-25:
Tried reflashing but it doesn't seem to have had any effect. Unfortunately, I've just got an Uno with me at the moment. I'll try the arduino->arduino test tomorrow when my project mate brings more. Thanks for your extremely responsive help!

ajfisher, 2017-01-25:
No worries. What version RPi are you using and I'll attempt to use the same.

hironroy, 2017-01-25:
Model B v1.1

ajfisher, 2017-01-25:
Try using i2cdetect -y 0 <<<<< Note the use of 0 here - I think there might
be a change in where the I2C connection is at between that board and the
newer ones

hironroy, 2017-01-25:
`Error: Could not open file `/dev/i2c-0' or `/dev/i2c/0': No such file or directory` I do have a `/dev/i2c-1` file

ajfisher, 2017-01-26:
@hironroy I've checked this all through with an RPi 3 and everything is working as expected. You can see steps on #71 that I went through.

Am going to try and find an older RPi to try this out on and will report back but certainly on a Model B V3 it's working correctly.

justbill2020, 2017-01-26:
i'm having this issue as well... i think this is the 1/11/2017 latest version of raspbian possibly causing these issues... i noticed you stated you were using an older version of raspbian and an older version of node than i was in #71...

ajfisher, 2017-01-26:
Possibly something has changed in pigpio which is a dependency for raspi-io
that may be causing this. I doubt it's a node version issue given the stack
trace.

Ping @nebrius - you seeing any issues recently as a result of latest
raspbian?

nebrius, 2017-01-26:
I actually haven't installed or tested the latest version yet (I missed that it came out). It could be that the way this is configured has changed though.

I would recommend running `sudo raspi-config`, go into advanced, and try enabling I2C that way (don't forget to reboot!). This should help us figure out if there's some new configuration bits that are needed

justbill2020, 2017-01-26:
i did do that already... i just wasn't sure because they did make changes to how SSH works...

hironroy, 2017-01-27:
So, I reflashed my SD Card. And with a fresh install. Followed [Enable I2C on Pi Tut](https://learn.adafruit.com/adafruits-raspberry-pi-lesson-4-gpio-setup/configuring-i2c). And the flashed Uno shows up correctly now. I think my SD card had gotten corrupted. I've successfully gotten a Pi B+ and Pi 2 recognizing the Uno backpack.

ajfisher, 2017-01-30:
That's great news, yes I was trying this yesterday and was getting a node-gyp fail on Pi 2 as well but didn't have much time to debug further. Possibly try and older version of raspi-io as it may be an issue with wiringpi versions.

Will close this off as it looks like backpack detection is okay but feel free to comment or drop into gitter for any other debugging related to the raspi-io side of things (especially the raspi-io gitter)

closed 2017-01-30 by ajfisher

hironroy, 2017-02-01:
Cool. Thanks a ton! It turned out that wiringpi wasn't installed.

jesse-s, 2017-08-21:
I have the exact same issue with the installation of node-pixel with the interchange command. D2 lifted with 5V, and `ic2detect` does show a detected device (#42). I have flashed the Nano manually through the `grunt build` command and uploaded it through the Arduino IDE. Should this achieve the same as `interchange install git+https://github.com/ajfisher/node-pixel -a nano`?

Because Johnny Five still gives a Device or Firmware error after echo'ing `Connected /dev/ttyUSB0`.

ajfisher, 2017-08-21:
I haven't had a chance to implement the interchange protocol on the
backpack properly so this will fail.

Once it's flashed you'll need to use it as a backpack. J5 doesn't talk to
it directly, it talks to it over I2C via another device such as an Arduino
or a raspberry pi.

Eg:
https://github.com/ajfisher/node-pixel/blob/master/docs/johnnyfive-i2c.md

ajfisher, 2017-08-21:
> Thanks for the answer. I updated my comment above after I posted it.
> I did exactly that, and even tested with other firmware that did work. It
> seems the node-pixel firmware doesn't work for J5. I have the same setup as
> the link you included.

The I2C backpack will definitely not work with Johnny five, you need an
Arduino running firmata to talk over I2C and you talk to the Arduino
running firmata over Johnny five.

If you try to talk to the I2C backpack firmware directly, you'll get there
error you've described because you're sending it firmata commands and it's
expecting I2C messages.

jesse-s, 2017-08-21:
Oh my, was I wrong. Cheers for clearing that up!

jesse-s, 2017-08-21:
Still a bit confused after reading the docs. I have the following setup now: https://github.com/ajfisher/node-pixel/blob/master/docs/breadboard/i2c_backpack_multipin_bb.png

I don't want to use the Firmata version, because I'm powering 6 strips with 60 LEDs each on them. What exactly do I need to flash from the Raspberry to the Arduino? I am running the J5 code from the Raspberry in order to control them.

ajfisher, 2017-08-21:
Refer here: https://github.com/ajfisher/node-pixel/blob/master/docs/multipin-i2c.md

On the arduino - flash the I2C backpack firmware.

Wire up the arduino to the RPi using the I2C connections and add a single strip to `PIN 0` on the arduino and ensure you're working. The key configuration piece is an array of LED lengths passed in to the controller which tells you how long each strip is per pin. Each strip is connected from Digital PIN 0 outwards.

```
    strip = new pixel.Strip({
        board: this,
        controller: "I2CBACKPACK",
        color_order: pixel.COLOR_ORDER.GRB,
        strips: [ 60, ] // just one strip added
    });

.. rest of your code to do something
```

Once you have that reliably working, now add more:

```
    strip = new pixel.Strip({
        board: this,
        controller: "I2CBACKPACK",
        color_order: pixel.COLOR_ORDER.GRB,
        strips: [ 60, 60, 60, 60, 60, 60] // 6 lengths of 60 LEDs
    });

  ... code here
```

jesse-s, 2017-08-22:
Yes, that's what I meant earlier. After flashing the Arduino with the I2C backpack firmware, it is not possible to connect to the Arduino from the Raspberry with J5 because of the `Device or Firmware` error. Since there is no board flashed with Firmata, how would this be possible running code from a Raspberry?

ajfisher, 2017-08-22:
My apologies - I see where the issue is. You need to run `raspi-io` to act as the J5 IO-plugin. I didn't realise you hadn't done that. Please see here: https://www.npmjs.com/package/raspi-io

Once you've installed it and then set up like this:

```
const Raspi = require('raspi-io');
const five = require('johnny-five');
const board = new five.Board({
  io: new Raspi()
});
```

At that point when you connect to the board and set up your node-pixel controller it will understand to use the I2C connection _of the RasPI_ - and it should work for you.

Can you please raise a new ticket (as that will remind me) to include the links to that on the I2C example pages and generate an example? (Alternatively, PRs are hugely appreciated - especially for this stuff while it's fresh in your mind - and you can do it straight from GitHub - no need to git clone).