ajfisher/node-pixel #96: Support for DotStar led strips (APA102C)

aknoerig, 2017-02-24:
They seem to work very similar to the NeoPixels, but are controlled through SPI instead of I2C.
So I assume it's a bit of work to support them with node-pixel?

ajfisher, 2017-02-27:
It's actually more of an issue with Firmata than for node-pixel. As SPI potentially causes some blocking behaviour it is not been supported to date. To roll this feature out on the backpack should be relatively straight forward however at that point you're basically using a backpack to create I2C functionality to support SPI which feels a little bad... though that's certainly possible.

aknoerig, 2017-02-27:
I see, thanks for the explanation! I guess I'll go with a custom protocol for now.

soundanalogous, 2017-02-27:
SPI support for Firmata is in progress:

- [Initial proposal](https://github.com/firmata/protocol/blob/master/proposals/spi-proposal.md)
- [Prototype Arduino implementation](https://github.com/firmata/arduino/pull/341)
- [Prototype firmata.js implementation](https://github.com/firmata/firmata.js/compare/spi-alpha)

 Only thing slowing it down is to date I've had zero feedback on the proposal.

aknoerig, 2017-02-27:
Wow, looks great! Thanks for the hard work, Jeff. Hope it comes through soon.