ajfisher/node-pixel #100: Arduino Micro support?

kurai021, 2017-04-10:
I see that actually Interchange only support 3 types of Arduino boards, but what about Micro?. Could be useful for projects that require smallest spaces.

ajfisher, 2017-04-10:
@kurai021 So you should be able to build it directly from the source. Having said that if you want to test it out you should be able to extend the grunt config to add the new board type `micro` and then get it to build. If that builds then you should be able to install it using interchange without any issues at that point.

ajfisher, 2017-04-11:
So you should be able to install this simply with:

`interchange install node-pixel -a micro --firmata` 

Will close this off so please reopen if this is still an issue

closed 2017-04-11 by ajfisher

kurai021, 2017-04-11:
hi, well as you said me in Twitter, the Micro is a little problematic with connections but was possible to load firmata with Arduino IDE (no problem) and Interchange (one in four attempts maybe). Thank you.

ajfisher, 2017-04-11:
If you have the output of one of the times it *didn't* work that you can
post here, I can take a look and see if we can improve that.