ajfisher/node-pixel #103: Using with Beaglebone Black

ZachChap23, 2017-06-22:
I've been trying to use this library with the beaglebone I've used this library before with arduino Nanos loaded with Firmata and I really liked it. However, whenever I try to set it up I receive the error 
`>> /var/lib/cloud9/node_modules/node-pixel/lib/pixel.js:86
                if (firmata.firmware.name !== 'node_pixel_firmata.ino') {
                                    ^

TypeError: Cannot read property 'name' of undefined
    at Strip.Controllers.FIRMATA.initialize.value (/var/lib/cloud9/node_modules/node-pixel/lib/pixel.js:86:37)
    at new Strip (/var/lib/cloud9/node_modules/node-pixel/lib/pixel.js:478:14)
    at Board.<anonymous> (/var/lib/cloud9/Test/Johnny-Five-Neopixel-Test.js:14:13)
    at emitNone (events.js:72:20)
    at Board.emit (events.js:166:7)
    at nextTickCallbackWith0Args (node.js:489:9)`

I believe this is due to the beaglebone not actually using firmata, but the beaglebone-io plugin. 

This is my code (based mostly off of examples to get started):
`var five = require('johnny-five');
var BeagleBone = require('beaglebone-io');

pixel = require("node-pixel");

var board = new five.Board({ 
  io: new BeagleBone()
});
var strip = null;
console.log('done setup')

board.on("ready", function() {

    strip = new pixel.Strip({
        board: this,
        controller: "firmata",
        strips: [ {pin: 'P9_12', length: 16}, ], // this is preferred form for definition
        gamma: 2.8, // set to a gamma that works nicely for WS2812
    });

    strip.on("ready", function() {
        strip.color("#ff0000");
        strip.show();
    });
});`

Any pointers on how to fix this are appreciated, or let me know if I just have soemthign set up completely wrong.

ajfisher, 2017-06-29:
The reason for this is you need to use the i2c backpack as your controller not firmata.

Though I suspect you're trying to pulse off the BBB pin directly maybe? If that's the case then node-pixel doesn't directly support that though I reckon if there's a library we could wrap for ws2812 protocol then we might be able to do that relatively easily.

In the mean time, if you use a nano over I2C and then `I2CBACKPACK` as the controller then you'll be good.

ZachChap23, 2017-06-29:
Thank you for your quick response. I wasn't sure if the library had a way to do directly off the beaglebone, but the I2Cbackpack is a great solution as well.

closed 2017-06-29 by ZachChap23