ajfisher/node-pixel #111: LED brightness

josephclawrence, 2017-10-11:
Is there any way to control LED brightness in node-pixel, and if not is it potentially on the roadmap? Thank you!

ajfisher, 2017-10-11:
There's not a way that's baked into the code right now. The way I do it is to use the `color` library.

At the end of the day, all the firmware cares about is getting the right RGB value to display the pixel so really you want to be computing that on the node side.

So you can do this:

```
let color = Color('#7743CE');

color.lighten(0.5);

strip.color(color.string()); // strip will now be 50% lighter than original
```

Now NodePixel has correction in it this works fairly reliably.

scottsweb, 2018-08-27:
Would be very interested in having brightness control too. The method above with the `color` library works to an extent but the results are still not perfect.

che-effe, 2018-12-17:
Has there been any movement on this? It would be really useful to set a color value and change just brightness values, similar to the way that FastLED utilizes HSV.

ajfisher, 2018-12-17:
PRs for this definitely accepted. As you can use HSV on the color side it
should just be a case of converting and then bumping the brightness up or
down

Elmaranin, 2019-02-26:
Any news on this theme? Yes, you can control 'visible' brightness with color manipulations, but the main problem is in constant and maximum power consumption of attached strips. This is huge problem for me.

closed 2023-06-17 by ajfisher