ajfisher/node-pixel #115: support for Raspberry pi?

JordanKlaers, 2017-10-27:
Does this library support controlling a neopixel strip from a pi zero w? 
I have npm and node installed and I have used johnny five to control an rgb led, but I notice this library requires standard firmata and such.

ajfisher, 2017-10-27:
So the way I'd do this is to use a cheap nano or micro attached to the RPi
0 using the I2C bus. Alleviates a stack of timing issues with RPi gpio. For
$2 of board it provides lots of flexibility and simplicity (and you can run
hundreds of LEDs).

Look at the I2C backpack example for more info on how this works

JordanKlaers, 2017-10-31:
Ok, so something like this https://www.adafruit.com/product/292 
I notice i2c backpacks seem to be for lcd screens thought.
would that be what you are referring to? Or what would the name of the chip be exactly?
the code makes sense, just unsure on the chip
 exaclty >_<
Thank you so much for your help!

ajfisher, 2017-10-31:
Something like this is all you're after. Conceptually it's doing the same
thing as the adafruit one (though that's only for LCD screens)

https://www.google.com.au/search?q=arduino+nano&rlz=1C5CHFA_enAU738AU738&oq=arduino+nano&aqs=chrome..69i57j69i60l5.2047j0j4&sourceid=chrome&ie=UTF-8

umbertix, 2018-03-05:
Sorry to reopen this thread, but is this actually possible without the extra chip? I'm a programmer not really a electronic technician and I would like something that is pretty much plug and play :/

Is this the right library for me?

ajfisher, 2018-03-05:
If you're using a raspberry pi then you'll need to use an additional board
to make it work.

closed 2018-07-05 by ajfisher

JordanKlaers, 2018-07-05:
The data transmission to the strip is extremely slow. I wanted to control
over Bluetooth and was doing it all in JavaScript using this library.  It
was so slow I ended up using an arduino that had Bluetooth built in and
coded it in the arduino language and it was incredibly faster.
You will need the extra chip if you use this library with the pi

vedmant, 2018-08-07:
There are manuals that show it's possible to user Raspberry PI directly with led strips, https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/

ajfisher, 2018-08-07:
Yep. This uses a hack to output the signal on an audio pin which is a high
speed pin you have on RPI and can use DMA to do that.

The node pixel library isn't an RPi library, it's designed to work
primarily for I2C because that's the most portable method of control. The
technique linked whilst it works on an RPi won't work on a beagle bone or a
tessel or an Arduino.

If someone wanted to Port and then maintain a controller in node-pixel for
DMA for a specific target board then we can look at that, but I don't have
enough hours to dedicate to keeping specific boards in sync with custom
code for each one unfortunately (as much as I'd love to work on node-pixel
more).