ajfisher/node-pixel #116: Analog pins unusable while using node-pixel ?

cotejp, 2017-12-05:
I am using node-pixel with Johnny-Five (Firmata/Arduino). I am receiving distance data over analog pin 0. As soon as I stard using node-pixel, no more analog data comes in. If I comment out the node-pixel code, everything goes back to normal and works fine.

Is there anything to do to use analog pins while using node-pixel?

Thanks!

cotejp, 2017-12-05:
Hmm... It seems that by forcefully calling `board.io.reportAnalogPin(6, 1)` it fixes the problem. I'm still not sure why node-pixel turns off analog pin reporting. Or maybe there's something I don't understand...

ajfisher, 2020-10-18:
Closing this as stale as haven't been able to replicate. Please reopen if still an issue and can show code.

closed 2020-10-18 by ajfisher