ajfisher/node-pixel #120: updating strip occasional crashes: Unknown system error -121

JordanKlaers, 2018-04-23:
I am sending string commands over blue to the pi (Using Bleno) then updating the leds based off the bluetooth data. I am using raspiberrypi zero w and the arduino nano as an i2c backpack.

I am able to send bluetooth commands and the leds on the strip do change successfully using the nano as a backpack, but too many signals makes it crash.

The handler for receiving incoming signals is as ->

`var busy = false;
WriteOnlyCharacteristic.prototype.onWriteRequest = function(data, offset, withoutResponse, callback) {
	if (!busy) {
		var expectation = recieveData.handleTheData(result, LEDObject);
		Promise.resolve(expectation).then(()=> {
			busy = false;
			callback(this.RESULT_SUCCESS);
		})	
	} 
};`

recieveData.handleTheData takes the string command and the ledObject which has the strip object.
I created the functionality as a promise to resolve the updating up the led strip before processing new incoming data. I am using `strip.pixel(index).color("rgb(0,50,0)"); strip.show();` to update the leds. 

Originally I had no promises for handling incoming data and updating to the strip. I had to wait about 1 second between sending data to be sure it wouldnt crash. I added a promise to asynchronously handle updating and that improved the speed at which I could send more data. I added a promise around all of it (as the code above is showing) and it has really improved things but it still crashes.

I have read the error might be about an inability to write to the i2c pins, a communication issue. I believe my problem my be related to the incoming Bluetooth signals interrupting the board being updated (or vise-versa) and I am not sure what options there are to troubleshoot.

ajfisher, 2018-07-05:
@JordanKlaers Did you resolve this issue?

JordanKlaers, 2018-07-05:
Yes, I don't recall off the top of my head, would you like me to go back
and find out? I think I just needed to catch errors from the board or when
updating the strip

ajfisher, 2018-07-25:
Cool - if that's all it was then I reckon we close it out.

closed 2018-07-25 by ajfisher