ajfisher/node-pixel #123: mega 2560 + pro mini problem

vettorazi, 2018-06-20:
hey guys, i am struggling with this all day long.
I don't have a UNO board, so i am trying to use a mega as main board and a pro mini as backpack.
i installed the node_pixel_firmata in the mega board, connected the sda and scl pins in the pro mini board and uploaded the backpack firmware.

i connected the strip in the pin 6 on the backpack board but not seems to work! :(

what could be the problem?
i tested all examples and the codes of the readme session, but it's all the same:
```
1529526591074 Available /dev/cu.usbmodem3131  
1529526591098 Connected /dev/cu.usbmodem3131  
1529526594305 Repl Initialized  
>> Board ready, lets add light
Strip ready, let's go
```

no errors but no lights either

ajfisher, 2018-06-21:
There's an error I've just noticed in the docs. The circuit diagram is
correct but the document is not.

On an I2C backpack the pixel strip should start from pin 0 NOT pin 6. Try
that and I reckon it will work for you.

vettorazi, 2018-06-21:
Hey ajfisher!
Thanks for the reply! I was about to write here this discover! I pinned in
the 0 (rx) and worked!
Thanks!

commit 45e829b2 (2018-07-05): Fixes readme typo and closes #123
closed 2018-07-05 by ajfisher