ajfisher/node-pixel #126: Speed questions

vettorazi, 2018-07-02:
Hey guys,
i made an application with node.js, in a matrix of 43x4 (172 neopixels in total) and i felt the animations a little slow.  My animation's framerate were a bit slow compared to what i see on the web.
i rewrited my code in c+ in the arduino IDE and booted directly in the board (just to compare it) and seems to be waaay to fast than the node.js application.

**question:**
i am looping all the pixels in my application, on the server side and throwing to the board. It's seems that the time to change all the leds is about 2 seconds. How can i speed up my code?
I am guessing that the way of "node.js>arduino board> i2c backpack > strip" is too long for a fast framerate.

If i upload directly to a raspberry it's gonna be faster?

thanks!

ajfisher, 2018-07-04:
@vettorazi If you can post some code I may be able to help a little more but I'll try to describe some possible solutions based on what you've described as the behaviour above.

if you're using Node-Pixel using Firmata then there's two things to bear in mind. Firstly Firmata is doing a lot more work than just a bit of pixel code. That has an overhead that we can't get around.

The second point is that when you use Firmata you're connecting over a serial cable over USB. This has an UPPER limit baud rate of 57600 bits per second. Don't forget that this is bidirectional so if Firmata happens to be sending data back then you're going to drop that rate accordingly.

Assuming uni directional flow FROM node TO firmata, you're left with an upper limit of 7200 bytes per second.

To understand what is happening under the hood, a message to trigger the `show` latching event requires about 4 bytes of data to be transmitted. 

Now, to update a single pixel and then show it, it takes 9 bytes of information to set the pixel value.

So if you wanted to update a single pixel and then show it then  it will "cost" you 13 bytes to do so. 

based on our 7200 bytes per second that would give you about 500 frames per second. Not bad...

But that's small scale and only the theory.

Practically, what that means is that if you have a 172 neopixels and you update all of them in a frame you're going to send 172 x 9 bytes over the wire - 1548 bytes. Add your `show` and that's 1552 per frame.

At best, assuming everything was working perfectly, there was no interface, and firmata wasn't doing anything else, then you'd expect to get a maximum frame rate of about 4 frames per second. In practice there's some wastage in there so you'd never see it that fast. 

So the advice I have without seeing your code is this:

* Don't rewrite every single pixel - try to limit what you need to write to just the diffs.
* If you don't need to, don't blank out the whole strip and rewrite it (don't think of it like a display)
* Call `show()` as infrequently as possible
* your eyes will only see big shifts in colour. As such don't write small changes as it's just soaking up brandwidth that no one is going to see - this isn't a retina display ;)

ajfisher, 2018-07-04:
I should add - create a minimal version of your code and then post it here and I can have a look. I have plenty of neopixels so can happily test something out.

vettorazi, 2018-07-04:
Oh thanks so much for the attention @ajfisher !

I am trying to make a interactive wall of leds, with a html5 canvas. so all the colors are sended in an array to the server.
I did a 'mini' version of my code below:

`
var frame7= [A ARRAY WITH A BUNCH OF HEX COLORS]
var frame8=[A ARRAY WITH A BUNCH OF HEX COLORS]
var frame9= ['#350000','#350000','#350000','#350000','#260000','#270000','#270000','#1f0500','#170c00', '#170c00','#180b00','#080a00','#070d00','#070c00','#070c00','#002100','#002100','#002000','#001801','#000d0c','#000e0d','#000e0d','#001734','#001a37','#001a36','#001737','#00002d','#00002b','#00002b','#04002a','#11002b','#12002b','#12002d','#36000d','#39000b','#39000b','#38000a','#540000','#540000','#540000','#490000','#410000','#420000','#f78a20','#f78a20','#f78a20','#f78a1e','#feb816','#feb814','#feb814','#ffd313','#fff101','#fff101','#fff102','#ced927','#cada2b','#cada2b','#cadb31','#02a554','#01a652','#01a652','#42c194','#6ec8bf','#6fc8bf','#6fc8bf','#07aadb','#00addd','#00addd','#04afdf','#544ea4','#524fa3','#524fa3','#573e9a','#672c93','#672c93','#672c93','#b32670','#b62468','#b62468','#b92268','#ee1c27','#ee1d23','#ee1d23','#f13828','#f35925','#f25a23','#f78a20','#f78a20','#f78a20','#f78a1f','#ffb814','#feb814','#feb814','#ffd10f','#fff101','#fff101','#fff101','#d1d727','#cada2b','#cada2b','#ccd82f','#01a554','#01a652','#01a652','#41c296','#6fc8bf','#6fc8bf','#6fc8bf','#07abd9','#00addd','#00addd','#04adde','#554da5','#524fa3','#524fa3','#5a419c','#672c93','#672c93','#672c93','#b32670','#b62468','#b62468','#b92268','#ee1c27','#ee1c24','#ee1c24','#f13828','#f25a24','#f25a23','#f78a20','#f78a20','#f78a20','#f78a1f','#ffb814',

var animation=[frame1, frame2, frame3, frame4, frame5, frame6, frame7, frame8];

//Y FOR THE FRAMES I FOR THE 'IMAGE' OF THE FRAME:

            for(y=0;y<animation.length;y++){
            for (i=0;i<146;i++){
                strip.pixel(i).color(animation[y][i]);
            }
            strip.show();
        }
`

i hope that you understand and can help me :)
i liked the tip of the big shift of colours. but i don't know if this could be the reason.

ajfisher, 2018-07-05:
Okay. Based on the code above and some mulling on this today, I think I know what the issue is @vettorazi - note I haven't tested this so it's just speculation.

If this is a representation of your code, then I reckon your double for loop is running into synchronisation problems with with the serial connection.

Basically this:

```
for(y=0;y<animation.length;y++){
  for (i=0;i<146;i++){
    strip.pixel(i).color(animation[y][i]);
  }
  strip.show();
}
```

Will attempt to run as fast as it possibly can and as a result of that it will attempt to run `.show()` before the last one is finished and more importantly, it will attempt to start writing the new frame to the pixel buffer before it's finished rendering it. 

What I would suggest is to refactor your code to run something like this (pseudoish JS):

```
const FPS = 1; // change this as you validate it's working.
const FRAME_TIME = 1000 / FPS; 
const animation = [ [...frame1 vals...], [...frame2 vals...], [...frame1 vals...] ];

let current_frame = 0;

let intv = setInterval(() => {
  
  animation[current_frame].forEach((c, i) => {
    strip.pixel(i).colour(c);
  });

  current_frame = (current_frame + 1 >= animation.length) ? 0 : current_frame + 1;

  strip.show();

}, FPS);
```

This will lock your `strip` updates to a specific frame rate. Initially in this code, it will run just one time a second but you can speed it up as you test it and make sure it's stable.

I hope that helps.

vettorazi, 2018-07-05:
hmmm!!! Good idea! this might help! i will try that! Thank you so much @ajfisher ! =)

ajfisher, 2018-07-07:
No worries - I'm hoping this fixes your issue.

I'm going to close this out but if it continues, just reopen it

closed 2018-07-07 by ajfisher