ajfisher/node-pixel #133: First pixel is green when strip set to black

dunhamjared, 2018-09-11:
**Issue:** First pixel is green when strip set to black.

----

As pointed out by @archi in another project, could this be related to the inverted level shifter found in some ws2812 LEDs?

Archi also says it could be that that the idle signal is not inverted for inverted operation -> ws2812 sees high instead of low when there is no data (not sure where that's handled).

https://github.com/jgarff/rpi_ws281x/issues/33

ajfisher, 2018-09-11:
> jgarff/rpi_ws281x#33 <https://github.com/jgarff/rpi_ws281x/issues/33>

Why are you using a level shifter? If you're connected to an Arduino the
signal line still run off 3.3 or 5v without any issue.

ajfisher, 2018-09-15:
> I was not using an inverted level shifter. I reconfigured the wires and
> the issue went away.

That's good to know - how did you have them wired up (just to help if
someone else comes across the same issue at a later stage). Thanks for the
doc edits today as well :)