ajfisher/node-pixel #137: Unable to flash backpack

DanHartman, 2018-11-19:
This seems to have symptoms similar to gh-92.  
I'm using the `interchange` command to provision to nano boards.  The first stumbling block was [interchange issue 36](https://github.com/johnny-five-io/nodebots-interchange/issues/36).  The fix posted there does allow me to continue.

The second issue was also an interchange that seems to be a breaking change in a newer version of `serialport`.  Editing the `package.json` and rolling back the version of serialport to `4.0.7` fixed an error that either said `parsers.readline is not a function` or `serialport.close is not a function`.  There is a [bot generated PR](https://github.com/johnny-five-io/nodebots-interchange/pull/35) to update the newest serialport inside of interchange.  But I think it will take more than that to use the newer parser api correctly.

Cool, feeling like this might work.  But no.  Pin D2 is pulled high to 5V.
```
interchange install git+https://github.com/ajfisher/node-pixel -a nano

Retrieving manifest data from GitHub
Downloading hex file
found nano on port /dev/tty.wchusbserial1410
connected
reset complete.
flashing, please wait...
flash complete.
Can't configure device. Did you remember to set your backpack into config mode?
```

At last I tried the suggestion in gh-93 to run `grunt build` and then open the src with the arduino ide to compile and flash.  `backpack:8:25: error: ./lw_ws2812.h: No such file or directory`

I'm stumped.

ajfisher, 2018-12-03:
@DanHartman Apologies, the docs relating to the latter part of your issue there is unclear on my side.

Essentially the specific configuration piece in Node-Pixel hasn't been implemented because there isn't actually a method to configure it. The reason for this is I've tried to optimise memory for running the pixels rather than implementing the interchange client. As such, if you get to that point, you'll find that the backpack is installed and it available on the default I2C device and should work fine. You'll use pins 0-7 to link your strips onto and should be good to go from there.

DanHartman, 2018-12-27:
Thanks for the clarification @ajfisher 
I was able to flash a backpack using the source code in his repo and the Arduino IDE.

closed 2018-12-27 by DanHartman