ajfisher/node-pixel #141: Firmware development: having an issue with grunt build and grunt compile

che-effe, 2019-01-15:
Apologies if I'm missing something from the firmware dev notes, but running grunt build clears out the build directories, then trying to run grunt compile gives me `command not found` whey trying to run exec on `firmatta_<key>` . eg 👍 ```Running "exec:firmata_uno" (exec) task
>> /bin/sh: undefined: command not found```. Am I missing a simple step?

ajfisher, 2019-01-15:
I'm on my phone at the moment. I suspect you need to set the Arduino path
command into an alias. I believe I might have put a note about this in the
development read me.

I'm traveling at the moment but should be able to take a look in a few
hours.

ajfisher, 2019-01-21:
Did you get this sorted?

ajfisher, 2019-01-24:
> Still on it, I only had a few hours to mess with it in the past 7 days.

That's ok. Was just checking in to make sure all was ok.

che-effe, 2019-01-24:
Hi, I'm still having an issue, I'm following the firmware docs here https://github.com/ajfisher/node-pixel/tree/master/firmware/src. Is this the correct doc to be looking through?

che-effe, 2019-01-24:
I think I'm good now, we just got the correct path in bash profile as 
`export ARDUINO_PATH="/Applications/Arduino.app/Contents/MacOS/Arduino"`
This is not listed in the docs i mentioned above... should I have been looking elsewhere?