ajfisher/node-pixel #144: using two arduinos

vaaski, 2019-02-04:
i want to configure my 300 LED strip like seen in [/docs/breadboard/i2c_backpack_arduino_bb.png](https://raw.githubusercontent.com/ajfisher/node-pixel/master/docs/breadboard/i2c_backpack_arduino_bb.png)
i flashed the backpack.ino onto my nano and StandardFirmataPlus on my uno.
then i connected the uno to my computer and started the example from [here](https://github.com/ajfisher/node-pixel/blob/master/docs/johnnyfive-i2c.md)
it detected the board and said the strip was ready, but no lights.

i am not sure how to configure this exactly, can someone provide some help?

ajfisher, 2019-02-04:
Without a specific error condition it's hard to debug remotely but I would
check the following:

- Use standard firmata rather than standardfirmataplus
- Put NodePixelFirmata on the Uno and test that your LED strip works okay
- Make sure the Nano is correctly powered as is the strip - if you're
powering the nano and the strip all from the Uno you probably don't have
enough power
- Make sure you have common grounds across all three items (uno, nano,
strip)
- Make sure the strip DIN is connected to the nano D0 pin

vaaski, 2019-02-06:
Sorry for getting back so late and thank you for the quick response.
I've verified that my strip is working with nodepixelfirmata like you suggested.
And of course I'm using an external power supply, grounds are shared.

After testing it a bit (which i thought wasn't necessary since I'm using another strip with node-pixel already) I found out that the strip isn't really the RGBW type of strip I expected. Instead, it has white LEDs on every spot, with **either** a red, green or blue one. [This](https://imgur.com/a/hSRyogS) is it's best impression of `#FF0000`.

I'll try the duo-arduino approach with another strip and I'll edit this once I've made progress. Also I'd be more than willing to give you a couple bucks on paypal if I get my strips working :)

ajfisher, 2019-02-06:
No worries at all. That sounds like an interesting strip layout. Got a
model number and I can take a peek at the specs?

At the moment there isn't official rgbw support but it's being worked on
and may need some testing assistance with different strip types to land it.

vaaski, 2019-02-06:
The only thing it came in was this
[image]

Yeah I've just seen the issues about RGBW support and I think it isn't worth it in my case, both in the cost- and the bandwidth difference.

vaaski, 2019-02-06:
quick y/n question, i can power my arduino nano from the same 5v power supply as i'm powering my strips with, right?

vaaski, 2019-02-06:
I've tried it with standard firmata and another strip, still no luck.
this is my setup, if it helps.
[image]

oh and the first LED went green when i bumped into the cable.

vaaski, 2019-02-08:
hey @ajfisher, not trying to annoying, but i would like to get node-pixel working for my strip.
is there something obvious i'm doing wrong? maybe connect the nano in a different way?
this is my first project with hardware and i'm trying to learn as much as possible.

ajfisher, 2019-02-09:
Okay - so the SK6812 pixels aren't currently supported. There's some work
being done *right now* to create support for them as they use a slightly
different protocol and are 4 bytes in size.

The set up you have looks fine. Powering the nano and the strip off the
same 5V should be fine as well.

So you'll need to wait until the SK6812 code lands which is being actively
worked on now and it looks like you're a prime candidate to be testing it
too.

vaaski, 2019-02-10:
Sorry but the strip from my last photo is a WS2812B, 144LEDs/m RGB strip, found [here](https://www.amazon.de/dp/B01LSF4PYW).

vaaski, 2019-02-21:
:/

ajfisher, 2019-03-09:
Apologies - have been under the pump at work.

So that was a 2812 and it was skipping the colour along the stand? What
happens if you define say just 3 pixels in length and then cycle the
colours through it and move them between pixel 1 - 3 eg: RGB, BRG, GRB etc
then YMC, MCY, CYM and so on...

closed 2020-06-05 by vaaski