ajfisher/node-pixel #145: Development debugging from firmware

che-effe, 2019-02-05:
Hi, I'm trying to finish up my work with supporting RGBW pixels. I'm looking to get debug output from the board to help fix that last few bugs and have been having some difficulty.
I would love to output the values of things like `color_depth` to ensure that the value changes based on params being passed in on initialization.

ajfisher, 2019-02-06:
So you can switch on a debug flag in the library:

https://github.com/ajfisher/node-pixel/blob/master/firmware/src/libs/ws2812/ws2812.h#L16

And then basically you just 

```
#ifdef DEBUG
Serial.write(foo);
#endif
```

That will mean you can leave the debug stuff in there later too.

ajfisher, 2019-02-06:
Oh and then that will print it all out to the serial interface.

ajfisher, 2019-02-07:
> Ah yes, we did see that. And thank you for the prompt response! The catch
> is we're relying on that Serial for J5 communication, no?
> We were able to output to the Serial interface, but then unable to connect
> the board through J5. And vice versa. I'm trying to confirm that we can
> toggle between color_depth of 4 and 3 to enable a default, and I desire to
> be able to switch between the two on the fly. In general, if I can get
> output from the board while using it would be invaluable.

So the way I do this is typically use it in backpack mode then I talk to
the backpack via I2C from an arduino for the node-pixel side and then I can
use the serial port on the backpack to watch the debug statements.

It's a little bit of a pain but it means you're not hacking firmata to have
a software serialport and writing stuff out on that (which then means you
need eg a USB-Serial converter and you're reading off some digital pins).