ajfisher/node-pixel #148: NoWritablePortError

jjrchrds, 2019-05-28:
I'm using johnn-five and node-pixel.
I've used interchange to flash the arduino.
I used the johnny-five instantiation snippet to get started. 
I'm using an 8 led neopixel stick on pin 6.
When I run the script, I get the following error
`NoWritablePortError: Node Pixel FIRMATA controller requires IO that can write out`

I'm using a kuman uno r3.

coolchip, 2019-05-30:
Same here with a Nano.
What means this error?

coolchip, 2019-05-30:
Seems to be a problem with firmata@2.0.0. With firmata@1.0.0 it works

jjrchrds, 2019-06-01:
> Seems to be a problem with firmata@2.0.0. With firmata@1.0.0 it works

Can you explain how you fixed it?

coolchip, 2019-06-01:
Yes. I used the example 'Firmata instantiation' and not the johnny-five example. Then I installed expicit the version 1.0.0 ```npm install firmata@1.0.0```. Without the version you get version 2.0.0. 
johnny-five is also using firmata 2.0.0. 
It seems that something changed in 2.0.0 and node-pixel doesn't know that.

jjrchrds, 2019-06-01:
Works! Did the same thing.

ajfisher, 2019-06-02:
Thanks for that - I have been away for the last few weeks so will need to
take a look at this. Given firmata has had a major version change I expect
this will definitely break some behaviour. It might take a little while to
get through what is going on but will attempt that over the next week. If
you have some time for some testing that would be greatly appreciated as
well.

In the meantime, as a note if anyone sees this - please `npm install
firmata@1.0.0` and then install everything else and it should continue to
work.

jjrchrds, 2019-06-02:
You'll also need to swap out the firmata 2.0.0 with 1.0.0 in johnny-five.

ajfisher, 2019-06-02:
Leaving this open for the moment to remind me to make the changes

ajfisher, 2019-06-05:
@jjrchrds @coolchip so I have a branch called `j5-firmata-upg` if you want to test it out, I believe this should just work though you'll need to flash a more up to date version of firmata - which can be found i the build folder.

ajfisher, 2019-06-05:
If that tests fine then I'll get that published.

FranDias, 2019-07-16:
Just installed using that branch and it worked :shipit:

Pimentoso, 2019-09-22:
Any news about merging this fix?

blia, 2019-11-25:
Hey, @ajfisher. Looks like it works well. Maybe it's time to merge it :)

Thanks

cococow123, 2019-12-10:
Having the same issue. The link above for the new firmata did not work and gave an error when trying to flash the arduino with a 404. Are there any quick solutions to fix this error?

sebastianbayer, 2020-02-05:
What I did so far:
```bash

$ nvm use
Now using node v12.14.1 (npm v6.13.4)

$ yarn 
...

$ npm install -g nodebots-interchange
...
```

Where the package.json looks like:
```
...
"dependencies": {
    "node-pixel": "git+https://github.com/ajfisher/node-pixel.git#j5-firmata-upg"
  }
...
```

If I do the next step:

```bash
$ interchange install git+https://github.com/ajfisher/node-pixel.git#j5-firmata-upg -a uno --firmata
```

I get this error. 

```bash
Downloading firmware
Installing git+https://github.com/ajfisher/node-pixel.git#j5-firmata-upg from github
(node:15835) UnhandledPromiseRejectionWarning: Error: Unable to retrieve file
    at /Users/USER/.nvm/versions/node/v12.14.1/lib/node_modules/nodebots-interchange/lib/downloader.js:101:15
    at processTicksAndRejections (internal/process/task_queues.js:94:5)
    at async Downloader.get_file_from_github (/Users/USER/.nvm/versions/node/v12.14.1/lib/node_modules/nodebots-interchange/lib/downloader.js:93:12)
(node:15835) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
(node:15835) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

Maybe some more information how you guys got it to work would help. Just a step by step solution, because I struggle understanding things like «swap the firmata version» in johnny five etc.

sebastianbayer, 2020-02-05:
And if I try to use the  branch for installation I get the following error:
```bash
Downloading firmware
Installing git+https://github.com/ajfisher/node-pixel.git#j5-firmata-upg from github
(node:4785) UnhandledPromiseRejectionWarning: Error: Unable to retrieve file
    at axios.get.then.catch.err (/Users/sebastianbayer/.nvm/versions/node/v10.18.0/lib/node_modules/nodebots-interchange/lib/downloader.js:101:15)
    at process._tickCallback (internal/process/next_tick.js:68:7)
(node:4785) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
(node:4785) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

ajfisher, 2020-02-06:
Apologies all. Has been a very busy period at work so not much time for
much else.

So the reason that latest error is happening is that it isn't a valid
branch. The path isn't relative to the ajfisher/node-pixel as it's
currently a fork.

I should get a chance to look at this over the weekend and test out though.

Reading the thread, it looks like it should be a relatively straightforward
merge in but needs some testing on equipment to validate.

ajfisher, 2020-02-07:
Apologies on this. Had some pretty full on work commitments the last few months so projects have all taken a hit.

I'll review over the course of this weekend and do some testing and make sure it's all working then merge this in and publish.

ajfisher, 2020-02-07:
Also - I think there's a regression in `interchange` that is stopping branches from being able to be used on installs. Will check that out as well.

ajfisher, 2020-02-07:
Okay so I've patched `interchange` so installing against the branch as given above should now work again. If peeps can test this then that would be great. Outside of that, I'll retest this whole branch over the weekend and then merge it in if it's good.

jjrchrds, 2020-02-07:
Will give it a go tonight. Thanks!

nicholasc, 2020-02-07:
yea works here!

yourGuy, 2020-03-27:
I ran successfully :
 interchange install git+https://github.com/ajfisher/node-pixel#j5-firmata-upg -p COM3 -a uno --firmata
But when I try to run the johnny-five snippet I still get : NoWritablePortError

AdamTheLawyer, 2020-05-11:
Is there any update on the status of this? I'm hitting the same problem.

Explicitly installing Firmata 1.0.0 didn't help, and I can't seem to install the j5-firmata-upg version of the library; it errors out. Debug log is below.

I have a feeling I'm missing something basic about npm (I'm new to this tool kit). 

Any thoughts?

> 0 info it worked if it ends with ok
> 1 verbose cli [
> 1 verbose cli   'C:\\Program Files\\nodejs\\node.exe',
> 1 verbose cli   'C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js',
> 1 verbose cli   'install',
> 1 verbose cli   'ajfisher/node-pixel#j5-firmata-upg'
> 1 verbose cli ]
> 2 info using npm@6.14.4
> 3 info using node@v12.16.3
> 4 verbose config Skipping project config: C:\Users\aklei/.npmrc. (matches userconfig)
> 5 verbose npm-session d42f0911ff03debb
> 6 silly install loadCurrentTree
> 7 silly install readLocalPackageData
> 8 silly fetchPackageMetaData error for github:ajfisher/node-pixel#j5-firmata-upg Command failed: git submodule update -q --init --recursive
> 8 silly fetchPackageMetaData warning: templates not found in C:\Users\aklei\AppData\Local\Temp\pacote-git-template-tmp\git-clone-b23adfc3
> 8 silly fetchPackageMetaData Host key verification failed.
> 8 silly fetchPackageMetaData fatal: Could not read from remote repository.
> 8 silly fetchPackageMetaData
> 8 silly fetchPackageMetaData Please make sure you have the correct access rights
> 8 silly fetchPackageMetaData and the repository exists.
> 8 silly fetchPackageMetaData fatal: clone of 'git@github.com:firmata/arduino.git' into submodule path 'C:/Users/aklei/AppData/Roaming/npm-cache/_cacache/tmp/git-clone-a5ceea43/firmware/src/libs/firmata/arduino' failed
> 8 silly fetchPackageMetaData Failed to clone 'firmware/src/libs/firmata/arduino'. Retry scheduled
> 8 silly fetchPackageMetaData warning: templates not found in C:\Users\aklei\AppData\Local\Temp\pacote-git-template-tmp\git-clone-b23adfc3
> 8 silly fetchPackageMetaData Host key verification failed.
> 8 silly fetchPackageMetaData fatal: Could not read from remote repository.
> 8 silly fetchPackageMetaData
> 8 silly fetchPackageMetaData Please make sure you have the correct access rights
> 8 silly fetchPackageMetaData and the repository exists.
> 8 silly fetchPackageMetaData fatal: clone of 'git@github.com:firmata/arduino.git' into submodule path 'C:/Users/aklei/AppData/Roaming/npm-cache/_cacache/tmp/git-clone-a5ceea43/firmware/src/libs/firmata/arduino' failed
> 8 silly fetchPackageMetaData Failed to clone 'firmware/src/libs/firmata/arduino' a second time, aborting
> 9 timing stage:rollbackFailedOptional Completed in 1ms
> 10 timing stage:runTopLevelLifecycles Completed in 2602ms
> 11 verbose stack Error: Command failed: git submodule update -q --init --recursive
> 11 verbose stack warning: templates not found in C:\Users\aklei\AppData\Local\Temp\pacote-git-template-tmp\git-clone-b23adfc3
> 11 verbose stack Host key verification failed.
> 11 verbose stack fatal: Could not read from remote repository.
> 11 verbose stack
> 11 verbose stack Please make sure you have the correct access rights
> 11 verbose stack and the repository exists.
> 11 verbose stack fatal: clone of 'git@github.com:firmata/arduino.git' into submodule path 'C:/Users/aklei/AppData/Roaming/npm-cache/_cacache/tmp/git-clone-a5ceea43/firmware/src/libs/firmata/arduino' failed
> 11 verbose stack Failed to clone 'firmware/src/libs/firmata/arduino'. Retry scheduled
> 11 verbose stack warning: templates not found in C:\Users\aklei\AppData\Local\Temp\pacote-git-template-tmp\git-clone-b23adfc3
> 11 verbose stack Host key verification failed.
> 11 verbose stack fatal: Could not read from remote repository.
> 11 verbose stack
> 11 verbose stack Please make sure you have the correct access rights
> 11 verbose stack and the repository exists.
> 11 verbose stack fatal: clone of 'git@github.com:firmata/arduino.git' into submodule path 'C:/Users/aklei/AppData/Roaming/npm-cache/_cacache/tmp/git-clone-a5ceea43/firmware/src/libs/firmata/arduino' failed
> 11 verbose stack Failed to clone 'firmware/src/libs/firmata/arduino' a second time, aborting
> 11 verbose stack
> 11 verbose stack     at ChildProcess.exithandler (child_process.js:303:12)
> 11 verbose stack     at ChildProcess.emit (events.js:310:20)
> 11 verbose stack     at maybeClose (internal/child_process.js:1021:16)
> 11 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
> 12 verbose cwd C:\Users\aklei\scripts\Johnny5
> 13 verbose Windows_NT 10.0.18363
> 14 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "ajfisher/node-pixel#j5-firmata-upg"
> 15 verbose node v12.16.3
> 16 verbose npm  v6.14.4
> 17 error code 1
> 18 error Command failed: git submodule update -q --init --recursive
> 18 error warning: templates not found in C:\Users\aklei\AppData\Local\Temp\pacote-git-template-tmp\git-clone-b23adfc3
> 18 error Host key verification failed.
> 18 error fatal: Could not read from remote repository.
> 18 error
> 18 error Please make sure you have the correct access rights
> 18 error and the repository exists.
> 18 error fatal: clone of 'git@github.com:firmata/arduino.git' into submodule path 'C:/Users/aklei/AppData/Roaming/npm-cache/_cacache/tmp/git-clone-a5ceea43/firmware/src/libs/firmata/arduino' failed
> 18 error Failed to clone 'firmware/src/libs/firmata/arduino'. Retry scheduled
[... cut ...]
> 18 error Host key verification failed.
> 18 error fatal: Could not read from remote repository.
> 18 error
> 18 error Please make sure you have the correct access rights
> 18 error and the repository exists.
> 18 error fatal: clone of 'git@github.com:firmata/arduino.git' into submodule path 'C:/Users/aklei/AppData/Roaming/npm-cache/_cacache/tmp/git-clone-a5ceea43/firmware/src/libs/firmata/arduino' failed
> 18 error Failed to clone 'firmware/src/libs/firmata/arduino' a second time, aborting
> 19 verbose exit [ 1, true ]

toolbox9, 2020-06-13:
First off, thanks for supporting this library.  It looks like an incredible amount of work. I am encountering this same issue and have tried to resolve it in every possible fashion described above to no avail.

I've tried installing firmata 1.0.0 before/after the other packages and altering the node version (because recent versions of node rejected the deprecated firamata package) with the exact same error message described above.

Can anyone provide insight into the exact steps they took to resolve the incompatibility?

Thank you so much.

```                    throw err;
                    ^

Error [NoWritablePortError]: Node Pixel FIRMATA controller requires IO that can write out
    at Strip.value (/Users/toolbox9/Desktop/test7/node_modules/node-pixel/lib/pixel.js:97:31)
    at new Strip (/Users/toolbox9/Desktop/test7/node_modules/node-pixel/lib/pixel.js:479:14)
    at Board.<anonymous> (/Users/toolbox9/Desktop/test7/strip.js:10:11)
    at Board.emit (events.js:327:22)
    at Board.EventEmitter.emit (domain.js:482:12)
    at processTicksAndRejections (internal/process/task_queues.js:79:11)
```

toolbox9, 2020-06-14:
I'm using Mac OSX Catalina 10.15.5 if that's of any clue.
I'm prepared to create my own library for animations, etc, but I'm having difficulty with the most simple of calls to the neopixel strip. Can anyone point me to any other node-based documentation to write the appropriate calls to adjust individual pixels?

toolbox9, 2020-06-18:
Thank you for generating this library and supporting it. Unfortunately, I was unable to overcome this issue despite attempting all of the above. I'd love to hear if anyone else tried something that worked. 

In the meantime, I was able to complete my project by switching to Python with this very simple library. Worked immediately. https://github.com/mdaffin/PyNeoPixel

manuman94, 2020-07-20:
Hi! Thank you for giving us this awesome library. 

Any advances in this issue? I'm at the same point and Node Firmata 1.0.0 is not working because of a serialport module failure.

Thank you, greetings.

mg-snake, 2020-08-23:
Hi -- is there any update on this?

Error [NoWritablePortError]: Node Pixel FIRMATA controller requires IO that can write out

I tried using v1.0.0 but I must be doing something wrong

manuman94, 2020-08-23:
If someone is still stuck at this:
If you are using Johnny Five check the Johnny Five folder at node_modules and check its own node_modules folder. Maybe there's inside a firmata v2 build. If you replace that with the 1.0 version it will work.

This works for other libraries that have firmata as a dependency.

mg-snake, 2020-08-23:
> If someone is still stuck at this:
> If you are using Johnny Five check the Johnny Five folder at node_modules and check its own node_modules folder. Maybe there's inside a firmata v2 build. If you [...]

thanks, I'll try that. not even sure if npm install firmata@1.0.0 --save is working for me. I don't see a firmata folder in my project node_modules folder.

manuman94, 2020-08-23:
> thanks, I'll try that. not even sure if npm install firmata@1.0.0 --save is working for me. I don't see a firmata folder in my project node_modules folder.

Hi! I'm not telling you to set the node_modules/firmata to the v1, I'm telling you to set the node_modules/johnny-five/node_modules/firmata to the v1. Maybe I didn't explain myself properly.

Greetings.

mg-snake, 2020-08-23:
> Hi! I'm not telling you to set the node_modules/firmata to the v1, I'm telling you to set the node_modules/johnny-five/node_modules/firmata to the v1. Maybe I didn't explain myself properly. [...]

Sorry -yes. I did both. running npm install firmata@1.0.0 --save inside the node_modules/johnny-five/node_modules folder

I just am not even seeing a firmata folder or anything. Think I am doing something wrong. :( if anyone has some time I can show on discord notsosolidsnake#0666 and maybe it will make more sense. It just seems like firmata is not saving when I run that

monteslu, 2020-08-24:
#163 likely fixes this.  I was able to get things working with the latest johnny-five with the latest firmata on that PR.

mg-snake, 2020-09-05:
Just circling back as I finally got this to work. I had to downgrade my version of node and then I was able to install firmata @ 1.0.0 and all is well.

ajfisher, 2020-10-04:
Howdy all - massive apologies on the tardiness of all of this. First day in many months I've had enough headspace to actually sit down and look at a code repo that isn't work related. 

Thank you @monteslu for the patch and Firmata lib knowledge.

#165  should address this and bring everything back up into line with latest versions of J5 and Firmata (and clean up a bunch of security things in the background too).

If this doesn't address it, please reopen.

closed 2020-10-04 by ajfisher