ajfisher/node-pixel #149: NeoPixel 16 RGB Ring Inconsistencies

jakowenko, 2019-06-09:
I have a NeoPixel Ring 16 LED ring that I'm controlling using node-pixel and johnny-five.

On the Uno all the LEDs show solid colors fine. As soon as I tried to do the channel fade (https://github.com/ajfisher/node-pixel/blob/master/examples/channel_fade.js) I noticed that the first pixel doesn't work or sometimes get stuck or lags behind a little.

I then used the exact same code on my Leonardo and it works just fine. The issue described above doesn't happen. I also have uploaded some sample rainbow fading code directly to the Arduino and all the LEDs function correctly.

I've attached some images/video to show the issue. I'm new to the Arduino community and appreciate any feedback as to what the issue is.

Uno:

Image showing channel fade animation with stuck / not working LED. Second image shows I can then set all LEDs to solid color just fine - https://imgur.com/a/yLSeydp

Video showing simple animation that works fine on the same Uno - https://www.youtube.com/watch?v=8NSoa7DLvRU&feature=youtu.be

Leonardo

Same setup and code as Uno. Everything works as expected - https://youtu.be/NAr8JF3PnGk

jakowenko, 2019-06-10:
Tuning down the FPS does seem to have helped. Though I'm still not sure why the same code on the Uno and Leonardo produced different results.

ajfisher, 2019-06-10:
That's really odd behaviour I've never seen before.

Interesting that it's seemingly related to the frame rate. What FPS were
you using?

closed 2021-10-06 by jakowenko