ajfisher/node-pixel #151: Support dressing more than 1000 + leds

keaixx365, 2019-07-18:
@ajfisher I see from the source code that there is a limitation of 500 when using the I2CBACKPACK. Is there a way we can try to control addressable 1000 + leds?

ajfisher, 2019-07-18:
The only way to do that is user a different Arduino. Essentially, the limit
comes from the memory allocation of a nano. If you used say a mega or
similar then you can up that number quite comfortably.

keaixx365, 2019-07-18:
@ajfisher  did you try any other arduino which has bigger memory? Could you recommender one? And is we use a different arduino, should we change the max limitation in the pixel.js?

ajfisher, 2019-07-19:
I've used an Arduino Mega and I think I set the value at about 1000 or
something and it worked. You just change that max limitation and it should
be fine - it's just a check to stop people trying to address 10000 LEDs on
an Arduino Uno and then wonder why it doesn't work when it fails silently.
(you can imagine what that would do to GH issues).

peteruithoven, 2021-06-09:
Using I2CBACKPACK involves 2 microcontrollers, which one would you recommend switching out for a more powerful one?

ajfisher, 2021-06-09:
So as with most electronics stuff you get a bit of tradeoff happening here.

Single Arduino Mega and connect via firmata gives you more
individual pixels you can control but will end up with a lot of message
overhead due to the way firmata messages get encoded. This is more of a
problem if you're doing a lot of high speed / large number switching.

If however you're doing smaller amounts of transitions then I2C backpack is
probably a cheaper option and more stable...

closed 2023-06-17 by ajfisher