ajfisher/node-pixel #158: Arduino Mega = No Lights, No Errors

ColinTravis, 2020-01-27:
Hi there, I've flashed the current version of node-pixels onto a standard Arduino Mega and I've tried running my own custom code as well as the examples. If I run this code with the same wiring on an Uno, it works as expected and my LEDs light. If I run it off of a Mega, I receive no errors, but I also have no lights. Using standard Arduino code, I'm able to get the LEDs to light with the Mega, but just not with node-pixels.

ColinTravis, 2020-01-28:
For reference, I had my LED ring plugged into Pin 50 of the Mega. I'm testing other pins now. Some do light up on different pins.. are some reserved?

hweeks, 2020-11-01:
some pins are reserved. i will try and find the arduino pin reservation docs

closed 2023-06-17 by ajfisher