ajfisher/node-pixel #182: Can't install

colecci, 2020-10-13:
When I try to install with `interchange install git+https://github.com/ajfisher/node-pixel -a uno --firmata` I keep getting the following error:
```
Downloading firmware
Installing git+https://github.com/ajfisher/node-pixel from github
Flashing firmware to board
(node:5064) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'length' of undefined
    at /usr/local/lib/node_modules/nodebots-interchange/node_modules/avrgirl-arduino/lib/connection.js:18:16
    at /usr/local/lib/node_modules/nodebots-interchange/node_modules/avrgirl-arduino/lib/connection.js:65:23
    at /usr/local/lib/node_modules/nodebots-interchange/node_modules/avrgirl-arduino/lib/connection.js:194:51
(node:5064) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:5064) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

ajfisher, 2020-10-13:
I'll have a look at this, this evening.

However, in the meantime:

- what os are you using?
- is your uno detected in terms of usb serial port?
- if you're on windows, try passing the -p arg to specify the port

That error is indicative of your machine not being able to establish a
connection to the board. Basically, no connections come back as available
to write to.

Related to this, are you using an Arduino Uno and not something else (eg
nano) ? In which case you need to set the board type for the -a argument.

ennoaragon, 2020-10-13:
Good morning, I ran into the same error last night, and specifying the port did the trick. `interchange install git+https://github.com/ajfisher/node-pixel -a uno --firmata -p COM3`

ajfisher, 2020-10-13:
Yeah I have run into this when doing some nodebots workshops as well. It
seems a lot of newer Windows machines have some internal devices that
appear like com ports and so it tries to bind to the first it can find (eg
above might be COM1 or COM2) and you need to give it a nudge to push the
connection to the right port.

If anyone wants to make a little PR for an update to the docs to note that
then that would be appreciated, if not, will update tomorrow during my day.

closed 2020-10-14 by ajfisher

Nytelife26, 2020-10-20:
@colecci stated he was on MacOS. How do you think merging a PR that's just a docs update for windows users fixes this issue?

ajfisher, 2020-10-21:
@Nytelife26 not sure where the OS was specified? Regardless, I can't replicate this on MacOS which is what I use so without additional information. If you have additional info to add to this then that would be greatly appreciated and happy to reopen.

PR #183 (2020-10-14): docs: Updated readme to provide extra help for windows users. 
Fixes #182
commit 5e22e8af (2020-10-14): docs: Updated readme to provide extra help for windows users. Fixes #182 (#183)