ajfisher/node-pixel #245: Teensy support?

peteruithoven, 2021-05-31:
More a question than an issue, is support for any of the [Teensy](https://www.pjrc.com/teensy/) models on the roadmap? 
I noticed there are no teensy models included in this file: 
https://github.com/ajfisher/node-pixel/blob/master/firmware/boards.js

Would it be a matter of finding the right `arduino:avr:pro:cpu=16MHzatmega328` like line and adding that?

ajfisher, 2021-06-01:
The only reason Teensy's aren't on roadmap is due to the cost of shipping
them to Australia so I can test them. If you have one and can test it all
and it works then that should work okay to include it. You need to build
the right target file which you'd add and then making sure the target
firmata builds correctly but other than that it *should work*....

peteruithoven, 2021-06-01:
Small update:
I just ran through the steps mentioned in https://github.com/ajfisher/node-pixel/blob/master/firmware/src/README.md
Ran into this issue: https://github.com/ajfisher/node-pixel/issues/246
But was able to perform the actions manual (without Make).

I then went through all the following steps so that I could compile and load ino files to teensies:
https://www.pjrc.com/teensy/tutorial.html

I had to remove all other Servo libraries except for the teensy specific one. 

Now I'm running into this compilation error:
```
.../node-pixel/firmware/build/node_pixel_firmata/light_ws2812.cpp: In member function 'void WS2812::ws2812_sendarray_mask(uint8_t*, uint16_t, uint8_t, uint8_t*, uint8_t*)':
light_ws2812.cpp:158: error: impossible constraint in 'asm'
     );
      ^
impossible constraint in 'asm'
```

peteruithoven, 2021-06-23:
@ajfisher you're probably just busy, but I've also send you an email.