ajfisher/node-pixel #246: cp: cannot stat './firmware/src/libs/firmata/arduino/*.{h,cpp}': No such file or directory

peteruithoven, 2021-06-01:
I'm running into an issue when running `make build-firmata`:
```
rm -rf ./firmware/build/node_pixel_firmata
Creating firmata build files
mkdir ./firmware/build/node_pixel_firmata
cp ./firmware/src/libs/firmata/arduino/*.{h,cpp} ./firmware/build/node_pixel_firmata/
cp: cannot stat './firmware/src/libs/firmata/arduino/*.{h,cpp}': No such file or directory
Makefile:102: recipe for target 'build-firmata' failed
make: *** [build-firmata] Error 1
```
I made sure to install the arduino submodule using `git submodule update --init --recursive`.

Running `ll` on the files it's trying to copy they are found just fine:
```
$ ll ./firmware/src/libs/firmata/arduino/*.{h,cpp}                                    
-rw-rw-r-- 1 peteruithoven peteruithoven  49K jun  1 21:06 ./firmware/src/libs/firmata/arduino/Boards.h
-rw-rw-r-- 1 peteruithoven peteruithoven 5,6K jun  1 21:06 ./firmware/src/libs/firmata/arduino/FirmataConstants.h
-rw-rw-r-- 1 peteruithoven peteruithoven  19K jun  1 21:06 ./firmware/src/libs/firmata/arduino/Firmata.cpp
-rw-rw-r-- 1 peteruithoven peteruithoven 8,5K jun  1 21:06 ./firmware/src/libs/firmata/arduino/FirmataDefines.h
-rw-rw-r-- 1 peteruithoven peteruithoven 8,0K jun  1 21:06 ./firmware/src/libs/firmata/arduino/Firmata.h
-rw-rw-r-- 1 peteruithoven peteruithoven  16K jun  1 21:06 ./firmware/src/libs/firmata/arduino/FirmataMarshaller.cpp
-rw-rw-r-- 1 peteruithoven peteruithoven 2,4K jun  1 21:06 ./firmware/src/libs/firmata/arduino/FirmataMarshaller.h
-rw-rw-r-- 1 peteruithoven peteruithoven  17K jun  1 21:06 ./firmware/src/libs/firmata/arduino/FirmataParser.cpp
-rw-rw-r-- 1 peteruithoven peteruithoven 4,2K jun  1 21:06 ./firmware/src/libs/firmata/arduino/FirmataParser.h
```

I'm using zsh on elementary OS 5.1.7 Hera which is built on Ubuntu 18.04.4 LTS. 

I feel like I'm missing something stupid. 

I noticed I can actually run the line manually:
```
cp ./firmware/src/libs/firmata/arduino/*.{h,cpp} ./firmware/build/node_pixel_firmata/
```

ajfisher, 2021-06-08:
Apologies for tardiness on this. Possibly a permissions issue here to do
with something in make. Might be worth making a simple make recipe in the
Makefile and just see if it can correctly create the folder and copy an
individual file across. Can't think *why* this would be a problem but may
help identifying it.

peteruithoven, 2021-06-08:
Good idea. I now have a very simple example. Note the comments for when it doesn't work. 
``` makefile
FIRMWARE_DIR = ./firmware
BUILD_DIR = $(FIRMWARE_DIR)/build
SRC_DIR = $(FIRMWARE_DIR)/src
LIBS_DIR = $(SRC_DIR)/libs

# make the Firmata build process to copy the files to the right place
FIRMATA_DEST_DIR = $(BUILD_DIR)/node_pixel_firmata
# FIRMATA_FILES = $(LIBS_DIR)/firmata/arduino/*.{h,cpp} # < Doesn't work
# FIRMATA_FILES = $(LIBS_DIR)/firmata/arduino/Boards.h # Works
FIRMATA_FILES = $(LIBS_DIR)/firmata/arduino/Firmata.cpp # Works
# FIRMATA_FILES = $(LIBS_DIR)/firmata/arduino/* # works (with cp -r)

build-firmata:
	cp $(FIRMATA_FILES) $(FIRMATA_DEST_DIR)/
```

peteruithoven, 2021-06-08:
Even simpler:
Does not work:
``` makefile
build-firmata:
	cp ./firmware/src/libs/firmata/arduino/*.{h,cpp} ./firmware/build/node_pixel_firmata/ 
```
Works: 
``` makefile
build-firmata:
	cp ./firmware/src/libs/firmata/arduino/Firmata.cpp ./firmware/build/node_pixel_firmata/	
```

peteruithoven, 2021-06-08:
I was afraid it was maybe the shell that was executing the Makefile. 

After reading:
https://www.gnu.org/software/make/manual/make.html#Choosing-the-Shell-1

This actually seems to help:
``` makefile
SHELL = /usr/bin/zsh

build-firmata:
	cp ./firmware/src/libs/firmata/arduino/*.{h,cpp} ./firmware/build/node_pixel_firmata/
```
This also works:
``` makefile
SHELL = /bin/bash

build-firmata:
	cp ./firmware/src/libs/firmata/arduino/*.{h,cpp} ./firmware/build/node_pixel_firmata/
```

peteruithoven, 2021-06-08:
I think the `{h,cpp}` bit just isn't supported by the shell that's executing makefiles on my system.

Crude alternative that also works: 
``` makefile
build-firmata:
	cp ./firmware/src/libs/firmata/arduino/*.h ./firmware/build/node_pixel_firmata/
	cp ./firmware/src/libs/firmata/arduino/*.cpp ./firmware/build/node_pixel_firmata/
```

peteruithoven, 2021-06-08:
Relevant: https://stackoverflow.com/a/589300/1544200

ajfisher, 2021-06-09:
Thanks for that update. Interesting it's not supported. Happy to just
switch it to the multiple line version of the same thing - probably makes
it a little more readable as well as more portable.

If you want to update the makefile and raise a PR that would be wonderful!

PR #250 (2023-06-17): Makefile build-firmware fix
Fixes: #246 

The `*.{cpp,h}` syntax didn't seem supported by the shell that executes the Makefile on Ubuntu 18.04 (sh, not bash or zsh). So I replaced it for 2 simpler steps.
closed 2023-06-17 by ajfisher