ajfisher/node-pixel #247: Can't install node pixel

johanneshermans, 2021-06-04:
We've tried to install node-pixel on a mac to control a led-strip with J5. But we've got this error`code 1
npm ERR! path /Users/johanneshermans/Desktop/Devine/Integration/4/demo/test leds/p01-blink/node_modules/firmata/node_modules/serialport
npm ERR! command failed
npm ERR! command sh -c prebuild-install || node-gyp rebuild
npm ERR! CXX(target) Release/obj.target/serialport/src/serialport.o
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@7.1.2
npm ERR! gyp info using node@16.1.0 | darwin | x64
npm ERR! gyp info find Python using Python version 3.8.2 found at "/Applications/Xcode.app/Contents/Developer/usr/bin/python3"
npm ERR! (node:32860) [DEP0150] DeprecationWarning: Setting process.config is deprecated. In the future the property will be read-only.
npm ERR! (Use `node --trace-deprecation ...` to show where the warning was created)
npm ERR! gyp info spawn /Applications/Xcode.app/Contents/Developer/usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/Users/johanneshermans/.nvm/versions/node/v16.1.0/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/johanneshermans/Desktop/Devine/Integration/4/demo/test leds/p01-blink/node_modules/firmata/node_modules/serialport/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/johanneshermans/.nvm/versions/node/v16.1.0/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/johanneshermans/Library/Caches/node-gyp/16.1.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/Users/johanneshermans/Library/Caches/node-gyp/16.1.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/Users/johanneshermans/.nvm/versions/node/v16.1.0/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/Users/johanneshermans/Library/Caches/node-gyp/16.1.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/Users/johanneshermans/Desktop/Devine/Integration/4/demo/test leds/p01-blink/node_modules/firmata/node_modules/serialport',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! In file included from ../src/serialport.cpp:1:
npm ERR! In file included from ../src/./serialport.h:6:
npm ERR! In file included from ../../../../nan/nan.h:2884:
npm ERR! ../../../../nan/nan_typedarray_contents.h:34:43: warning: 'GetContents' is deprecated: Use GetBackingStore. See http://crbug.com/v8/9908. [-Wdeprecated-declarations]
npm ERR!       data   = static_cast<char*>(buffer->GetContents().Data()) + byte_offset;
npm ERR!                                           ^
npm ERR! /Users/johanneshermans/Library/Caches/node-gyp/16.1.0/include/node/v8.h:5571:3: note: 'GetContents' has been explicitly marked deprecated here
npm ERR!   V8_DEPRECATED("Use GetBackingStore. See http://crbug.com/v8/9908.")
npm ERR!   ^
npm ERR! /Users/johanneshermans/Library/Caches/node-gyp/16.1.0/include/node/v8config.h:415:35: note: expanded from macro 'V8_DEPRECATED'
npm ERR! # define V8_DEPRECATED(message) [[deprecated(message)]]
npm ERR!                                   ^
npm ERR! ../src/serialport.cpp:41:48: error: too few arguments to function call, single argument 'context' was not specified
npm ERR!   v8::String::Utf8Value path(info[0]->ToString());
npm ERR!                              ~~~~~~~~~~~~~~~~~ ^
npm ERR! /Users/johanneshermans/Library/Caches/node-gyp/16.1.0/include/node/v8.h:3019:3: note: 'ToString' declared here
npm ERR!   V8_WARN_UNUSED_RESULT MaybeLocal<String> ToString(
npm ERR!   ^
npm ERR! /Users/johanneshermans/Library/Caches/node-gyp/16.1.0/include/node/v8config.h:452:31: note: expanded from macro 'V8_WARN_UNUSED_RESULT'
npm ERR! #define V8_WARN_UNUSED_RESULT __attribute__((warn_unused_result))
npm ERR!                               ^
npm ERR! ../src/serialport.cpp:48:53: error: too few arguments to function call, single argument 'context' was not specified
npm ERR!   v8::Local<v8::Object> options = info[1]->ToObject();
npm ERR!                                   ~~~~~~~~~~~~~~~~~ ^
npm ERR! /Users/johanneshermans/Library/Caches/node-gyp/16.1.0/include/node/v8.h:3031:3: note: 'ToObject' declared here
npm ERR!   V8_WARN_UNUSED_RESULT MaybeLocal<Object> ToObject(
npm ERR!   ^
npm ERR! /Users/johanneshermans/Library/Caches/node-gyp/16.1.0/include/node/v8config.h:452:31: note: expanded from macro 'V8_WARN_UNUSED_RESULT'
npm ERR! #define V8_WARN_UNUSED_RESULT __attribute__((warn_unused_result))
npm ERR!                               ^
npm ERR! ../src/serialport.cpp:95:18: warning: 'Call' is deprecated [-Wdeprecated-declarations]
npm ERR!   data->callback.Call(2, argv);
[... 113 lines cut ...]
npm ERR!   void init(v8::Handle<v8::Object> target) {
npm ERR!             ~~~~^
npm ERR! ../src/serialport.cpp:413:36: error: use of undeclared identifier 'target'
npm ERR!   void init(v8::Handle<v8::Object> target) {
npm ERR!                                    ^
npm ERR! ../src/serialport.cpp:413:43: error: expected ';' after top level declarator
npm ERR!   void init(v8::Handle<v8::Object> target) {
npm ERR!                                           ^
npm ERR!                                           ;
npm ERR! 8 warnings and 12 errors generated.
npm ERR! make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/Users/johanneshermans/.nvm/versions/node/v16.1.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:365:28)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
npm ERR! gyp ERR! System Darwin 20.3.0
npm ERR! gyp ERR! command "/Users/johanneshermans/.nvm/versions/node/v16.1.0/bin/node" "/Users/johanneshermans/.nvm/versions/node/v16.1.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /Users/johanneshermans/Desktop/Devine/Integration/4/demo/test leds/p01-blink/node_modules/firmata/node_modules/serialport
npm ERR! gyp ERR! node -v v16.1.0
npm ERR! gyp ERR! node-gyp -v v7.1.2
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/johanneshermans/.npm/_logs/2021-06-04T15_45_25_098Z-debug.log`

ajfisher, 2021-06-05:
That looks like a serial port issue based on the output

closed 2021-06-07 by johanneshermans

PnsDev, 2021-06-30:
I'm having the same issue, is there a way to fix this? I'm on windows 10