ajfisher/node-pixel #248: Problem with  maximum leds

johanneshermans, 2021-06-07:
Hi, We are program a WS2812B led strip with 300 LED's, The problem is that we can't handle more than 216 pixel leds. We are working with arduino uno, J5 and firmata. Is there a solution?

ajfisher, 2021-06-07:
Unfortunately not. The reason for the limit is that every pixel takes 3
bytes in ram to represent and firmata takes up a bunch as well. That limit
was chosen after a lot of benchmarking to determine how much team was
likely to be in use that was volatile and then maximize the remainder for
the strip.

You can use a subordinate second Arduino of some form using I2C bus and
you'll get a lot more(1000) or you can use an Arduino mega and you can use
thousands.

Unfortunately this is a hardware limit

ScreamZ, 2023-01-26:
@ajfisher IS 216 The limit on a single strip ? Or on the whole strip that can be managed by a single UNO ? I imagine for a single uno ?

ajfisher, 2023-01-26:
@ScreamZ it is the total number of LEDs available on an uno (really atmega328) you can have before you start running into memory issues. If you run in backpack mode instead you can have about double that without too much drama.

ScreamZ, 2023-01-27:
Thanks for your work on that

closed 2023-06-17 by ajfisher