ajfisher/node-pixel #253: Usage with Arduino Mega

qMalte, 2021-06-29:
Hello all,

unfortunately I can't switch LEDs with the Arduino Mega at the moment.

I use NodePixel in combination with J5, when I output the Strip object I get the following hint:
`Strip { dep_warning: { stripLength: false, gammaValue: false } }`

Strangely enough it works after the third request, but the LED then also switches in the wrong color (red instead of white) and with the wrong number of LEDs.

My source code looks like this:

```
const strip = new pixel.Strip({
    data: port,
    length: 128,
    color_order: pixel.COLOR_ORDER.GRB,
    board: board,
    controller: "FIRMATA"
});
console.log(strip);
strip.on("ready", () => {
    console.log("Strip rdy!");
    strip.color([255, 255, 255]);
    strip.show();
});
```

I am looking forward to your feedback.

Many greetings

Malte

ajfisher, 2021-06-29:
Never seen this before. Are you sure you're using a ws2812 strip? Feel like
what you're describing might be a different strip type

qMalte, 2021-06-30:
Yes exactly, enclosed are the technical specifications:

Product Name: WS2812B LED Pixel Strip
Model: HC-F5V-60L-60LED-B
Working Voltage: DC5V
Quantity: 60LEDs / M (3.3ft)
Power: 18W / M (3.3ft)
Packing: 5M (16.4ft) / Roll
Connection: VCC: RED, DAT: GREEN, GND: WHITE

Many greetings

Malte

ajfisher, 2021-06-30:
Hmm interesting

Do you have another Arduino to hand, one that isn't a mega (eg uno or
something)?

Am just trying to isolate where the issue may be coming from. I don't have
a mega close by for a couple of days to test against.

qMalte, 2021-06-30:
Unfortunately I have no other Arduino model available here at the moment. Can it possibly be because I have not yet adjusted the length of the LED strip and this is still 5 meters long and contains 300 pixels, which are connected together?

Many greetings

Malte

ajfisher, 2021-07-01:
that shouldn't really impact it - if you set the length to 128 then the way
the strip and code works it will just use the first 128 of the 300 pixels...

Just out of interest - how are you powering those 300 LEDs?

closed 2023-06-17 by ajfisher