ajfisher/node-pixel #254: Error when installing node-pixel (serialport)

PnsDev, 2021-06-30:
# What's going on
Trying to install by doing `npm i node-pixel` returns an error
```
D:\GitHub\ledslightskms>npm i node-pixel
npm ERR! code 1
npm ERR! path D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport
npm ERR! command failed
npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c prebuild-install || node-gyp rebuild
npm ERR! Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
npm ERR!   serialport.cpp
npm ERR! ..\src\serialport.cpp(41): error C2660: 'v8::Value::ToString': function does not take 0 arguments [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR! ..\src\serialport.cpp(41): error C2512: 'v8::String::Utf8Value::Utf8Value': no appropriate default constructor available [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR! ..\src\serialport.cpp(48): error C2660: 'v8::Value::ToObject': function does not take 0 arguments [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR! ..\src\serialport.cpp(95): warning C4996: 'Nan::Callback::Call': was declared deprecated [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR!   D:\GitHub\ledslightskms\node_modules\nan\nan.h(1742): note: see declaration of 'Nan::Callback::Call'
npm ERR! ..\src\serialport.cpp(113): error C2660: 'v8::Value::ToObject': function does not take 0 arguments [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR! ..\src\serialport.cpp(150): warning C4996: 'Nan::Callback::Call': was declared deprecated [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR!   D:\GitHub\ledslightskms\node_modules\nan\nan.h(1742): note: see declaration of 'Nan::Callback::Call'
npm ERR! ..\src\serialport.cpp(188): warning C4996: 'Nan::Callback::Call': was declared deprecated [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR!   D:\GitHub\ledslightskms\node_modules\nan\nan.h(1742): note: see declaration of 'Nan::Callback::Call'
npm ERR! ..\src\serialport.cpp(231): warning C4996: 'Nan::Callback::Call': was declared deprecated [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR!   D:\GitHub\ledslightskms\node_modules\nan\nan.h(1742): note: see declaration of 'Nan::Callback::Call'
npm ERR! ..\src\serialport.cpp(250): error C2660: 'v8::Value::ToObject': function does not take 0 arguments [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR! ..\src\serialport.cpp(285): warning C4996: 'Nan::Callback::Call': was declared deprecated [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR!   D:\GitHub\ledslightskms\node_modules\nan\nan.h(1742): note: see declaration of 'Nan::Callback::Call'
npm ERR! ..\src\serialport.cpp(329): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR! ..\src\serialport.cpp(330): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR! ..\src\serialport.cpp(331): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR! ..\src\serialport.cpp(336): warning C4996: 'Nan::Callback::Call': was declared deprecated [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR!   D:\GitHub\ledslightskms\node_modules\nan\nan.h(1742): note: see declaration of 'Nan::Callback::Call'
npm ERR! ..\src\serialport.cpp(377): warning C4996: 'Nan::Callback::Call': was declared deprecated [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR!   D:\GitHub\ledslightskms\node_modules\nan\nan.h(1742): note: see declaration of 'Nan::Callback::Call'
npm ERR! ..\src\serialport.cpp(413): error C2039: 'Handle': is not a member of 'v8' [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR!   c:\users\user\appdata\local\node-gyp\cache\14.15.0\include\node\v8-platform.h(16): note: see declaration of 'v8'
npm ERR! ..\src\serialport.cpp(413): error C2065: 'Handle': undeclared identifier [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR! ..\src\serialport.cpp(413): error C2275: 'v8::Object': illegal use of this type as an expression [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR!   ..\src\serialport.cpp(413): note: see declaration of 'v8::Object'
npm ERR! ..\src\serialport.cpp(413): error C2065: 'target': undeclared identifier [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR! ..\src\serialport.cpp(413): error C2182: 'init': illegal use of type 'void' [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR! ..\src\serialport.cpp(413): error C2448: 'init': function-style initializer appears to be a function definition [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR! ..\src\serialport.cpp(437): warning C4312: 'type cast': conversion from 'int' to 'node::addon_register_func' of greater size [D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport\build\serialport.vcxproj]
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@8.1.0
npm ERR! gyp info using node@14.15.0 | win32 | x64
npm ERR! gyp info find Python using Python version 3.9.0 found at "C:\Python39\python.exe"
npm ERR! gyp info find VS using VS2015 (14.0) found at:
npm ERR! gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio 14.0"
npm ERR! gyp info find VS run with --verbose for detailed information
npm ERR! gyp info spawn C:\Python39\python.exe
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   'D:\\GitHub\\ledslightskms\\node_modules\\node-gyp\\gyp\\gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'msvs',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'D:\\GitHub\\ledslightskms\\node_modules\\firmata\\node_modules\\serialport\\build\\config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'D:\\GitHub\\ledslightskms\\node_modules\\node-gyp\\addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'C:\\Users\\user\\AppData\\Local\\node-gyp\\Cache\\14.15.0\\include\\node\\common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=C:\\Users\\user\\AppData\\Local\\node-gyp\\Cache\\14.15.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=D:\\GitHub\\ledslightskms\\node_modules\\node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=C:\\\\Users\\\\user\\\\AppData\\\\Local\\\\node-gyp\\\\Cache\\\\14.15.0\\\\<(target_arch)\\\\node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=D:\\GitHub\\ledslightskms\\node_modules\\firmata\\node_modules\\serialport',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
[... 2 lines cut ...]
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   'build/binding.sln',
npm ERR! gyp info spawn args   '/clp:Verbosity=minimal',
npm ERR! gyp info spawn args   '/nologo',
npm ERR! gyp info spawn args   '/p:Configuration=Release;Platform=x64'
npm ERR! gyp info spawn args ]
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe` failed with exit code: 1
[... cut ...]
npm ERR! gyp ERR! System Windows_NT 10.0.19043
npm ERR! gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "D:\\GitHub\\ledslightskms\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd D:\GitHub\ledslightskms\node_modules\firmata\node_modules\serialport
npm ERR! gyp ERR! node -v v14.15.0
npm ERR! gyp ERR! node-gyp -v v8.1.0
npm ERR! gyp ERR! not ok
```
# Other Info
OS: Windows 10
Node Ver: 14.15.0
Npm Ver: 7.18.1 
I think it's a serial port issue but I'm not sure how to fix it.

ajfisher, 2021-07-01:
> npm ERR! gyp info using ***@***.*** | win32 | x64

To me this looks like an error installing a specific version of serial port
and not being able to get the binary version so it's trying to build it
from scratch and failing.

If you can try a couple of things:

   -  Between each go of this - remove your node-modules folder so you're
   operating from a clean slate.
   - Try and just do an npm install serialport with nothing else installed
   (or in a blank folder) and see if you get the same error - that suggests a
   general build error.
   - Try an install node-pixel from latest github version npm install git+
   https://github.com/ajfisher/node-pixel (I don't publish every time a
   package update occurs so might be just a tiny bit of drift on supported
   upstream packages)

Let me know how the results of that goes.

PnsDev, 2021-07-03:
I tried all the suggestions and none of them worked for me. 
When attempting to install serialport it installed fine as well.

ajfisher, 2021-07-05:
Hmm - that's really odd - I've installed it directly using nvm to get version 14.15.0 but I'm on a Mac and the serialport dependency just downloads from a binary.

Can I suggest using NVM to "downgrade" to an earlier 14.x version and just see what happens?

Also try and install firmata directly as it's a peer dependency and won't necessarily be installed. ie `npm install firmata` and see if that works okay (might be that it's that specific dependency link that's the issue on windows).

closed 2021-07-13 by PnsDev

Workshop2, 2021-08-01:
I am also encountering this. I have completely re-installed vs2017 VC++ and all that stuff and still can't get past this.

When playing with an empty repo, it seems the older versions of npm firmata fail to install. From v2 npm seems happy. 

I have also installed serialport directly and it seems fine.

Any chance the deps can be bumped to fix this issue?