ajfisher/node-pixel #259: Issue flashing Firmata to Arduino Board

danielduan2002, 2021-07-08:
I successfully installed johnny-five as well as the node-pixel and nodebots-interchange packages. After copying the command to flash my Arduino with the custom Firmata, I received these two lines after a brief wait
```
Downloading firmware

Flashing firmware to board
```

However, after proceeding and running the node app, this error gets spit out.
```
(node:19580) UnhandledPromiseRejectionWarning: IncorrectFirmataVersionError: Please upload NodePixel Firmata to the board
```

I'm not sure where the issue lies exactly, so any help or advice is greatly appreciated.

Specs:
Windows 10
Node v14.17.3
npm v6.14.13
Arduino Uno Rev3

ajfisher, 2021-07-08:
What version of J5 are you using? I suspect with the recent release this is
going to cause an issue with the version of firmata it's expecting to pull
through.

danielduan2002, 2021-07-08:
I tried downgrading to johnny-five@2.0.0, but I'm still receiving the same error message.

I copied my node modules below if that might be of any help.
```
+-- firmata@0.19.1
| +-- browser-serialport@2.1.0
| `-- serialport@6.0.5
|   +-- bindings@1.3.0
|   +-- commander@2.20.3
|   +-- debug@3.2.7
|   | `-- ms@2.1.2
|   +-- nan@2.14.2
|   +-- prebuild-install@2.5.3
|   | +-- detect-libc@1.0.3
|   | +-- expand-template@1.1.1
|   | +-- github-from-package@0.0.0
|   | +-- minimist@1.2.5
|   | +-- mkdirp@0.5.5
|   | | `-- minimist@1.2.5 deduped
|   | +-- node-abi@2.30.0
|   | | `-- semver@5.7.1
|   | +-- noop-logger@0.1.1
|   | +-- npmlog@4.1.2
|   | | +-- are-we-there-yet@1.1.5
|   | | | +-- delegates@1.0.0
|   | | | `-- readable-stream@2.3.7 deduped
|   | | +-- console-control-strings@1.1.0
|   | | +-- gauge@2.7.4
|   | | | +-- aproba@1.2.0
|   | | | +-- console-control-strings@1.1.0 deduped
|   | | | +-- has-unicode@2.0.1
|   | | | +-- object-assign@4.1.1
|   | | | +-- signal-exit@3.0.3
|   | | | +-- string-width@1.0.2
|   | | | | +-- code-point-at@1.1.0
|   | | | | +-- is-fullwidth-code-point@1.0.0
|   | | | | | `-- number-is-nan@1.0.1
|   | | | | `-- strip-ansi@3.0.1 deduped
|   | | | +-- strip-ansi@3.0.1
|   | | | | `-- ansi-regex@2.1.1
|   | | | `-- wide-align@1.1.3
|   | | |   `-- string-width@1.0.2 deduped
|   | | `-- set-blocking@2.0.0
|   | +-- os-homedir@1.0.2
|   | +-- pump@2.0.1
|   | | +-- end-of-stream@1.4.4
|   | | | `-- once@1.4.0 deduped
|   | | `-- once@1.4.0
|   | |   `-- wrappy@1.0.2
|   | +-- rc@1.2.8
|   | | +-- deep-extend@0.6.0
|   | | +-- ini@1.3.8
|   | | +-- minimist@1.2.5 deduped
|   | | `-- strip-json-comments@2.0.1
|   | +-- simple-get@2.8.1
|   | | +-- decompress-response@3.3.0
|   | | | `-- mimic-response@1.0.1
|   | | +-- once@1.4.0 deduped
|   | | `-- simple-concat@1.0.1
|   | +-- tar-fs@1.16.3
|   | | +-- chownr@1.1.4
|   | | +-- mkdirp@0.5.5 deduped
|   | | +-- pump@1.0.3
|   | | | +-- end-of-stream@1.4.4 deduped
|   | | | `-- once@1.4.0 deduped
|   | | `-- tar-stream@1.6.2
|   | |   +-- bl@1.2.3
|   | |   | +-- readable-stream@2.3.7 deduped
|   | |   | `-- safe-buffer@5.1.2 deduped
|   | |   +-- buffer-alloc@1.2.0
[... 280 lines cut ...]
| | |   |   `-- readable-stream@3.6.0
| | |   |     +-- inherits@2.0.4 deduped
| | |   |     +-- string_decoder@1.1.1 deduped
| | |   |     `-- util-deprecate@1.0.2 deduped
| | |   +-- tunnel-agent@0.6.0 deduped
| | |   `-- which-pm-runs@1.0.0 deduped
| | +-- @serialport/parser-byte-length@8.0.6
| | +-- @serialport/parser-cctalk@8.0.6
| | +-- @serialport/parser-delimiter@8.0.6
| | +-- @serialport/parser-readline@8.0.6
| | | `-- @serialport/parser-delimiter@8.0.6 deduped
| | +-- @serialport/parser-ready@8.0.6
| | +-- @serialport/parser-regex@8.0.6
| | +-- @serialport/stream@8.0.6
| | | `-- debug@4.3.2 deduped
| | `-- debug@4.3.2
| |   `-- ms@2.1.2 deduped
| `-- temporal@0.7.1
+-- node-pixel@0.11.0
| `-- color-string@1.5.5
|   +-- color-name@1.1.4
|   `-- simple-swizzle@0.2.2
|     `-- is-arrayish@0.3.2
`-- nwjs-j5-fix@1.0.6
```

danielduan2002, 2021-07-10:
Update: In the process of debugging this error, I pulled this project directly and that fixed the issue of firmata not flashing onto the Arduino board. My best guess as to what was causing the error in the first place was conflicting versions of node modules that node-pixel shared with the initial project that I had tried to install node-pixel into earlier. 

During the process of debugging, I also ran into a related issue that I notice has been brought up in other threads but not resolved officially. If anyone is running into an error during the initial flashing that says:
```
Error: Sending 3020: receiveData timeout after 400ms.
```
What caused this issue for me is I had a peripheral module connected to my board (specifically a MIDI Shield for the Arduino Uno), which occupies the Rx and Tx pins during its on state. To upload code to the board or flash it with firmware, those pins need to be free. 

Ajfisher, thank you for the prompt response to my issue and your advice in solving it. In the end the cause turned out to be my own incompetence, but I appreciate the advice and help!

closed 2021-07-10 by danielduan2002