ajfisher/node-pixel #285: Board

aronvandervalk, 2022-01-13:
Hi!

I'm trying to get my Pi fired up with the node-pixel library, together with I2C backpack. 
I installed the packages and try to run this simple code:

`var five = require("johnny-five");
var pixel = require("node-pixel");

var opts = {};
opts.port = process.argv[2] || "";

console.dir(process.argv[2]);

var board = new five.Board(opts);
var strip = null;

var fps = 20; // how many frames per second do you want to try?

board.on("ready", function() {

    console.log("Board ready, lets add light");
});`

But I keep getting the result after running this code:
`1642108712270 Board Looking for connected device
1642108720067 Board No connected device found`

It seems it cannot find "node" as the board?

Also, when I try to log `process.argv[2]` I get an undefined, as it does not exists.

ajfisher, 2022-01-13:
`process.argv[2]` here is an explicit path to the board in question.

This script that you're using here isn't really how you'd run this because
you're not running a firmata based board, you're using an I2C backpack.

As a result, you want to use Raspi-io to provide the board interface to
johnny five and then you should be good to go from there.

Have at this example for a simple LED Blink and the basic structure for the
set up: http://johnny-five.io/examples/raspi-io/

Assuming that works, then just replace the board.ready event with the node
pixel stuff and you should get some coloured LEDs.

aronvandervalk, 2022-01-13:
Thanks for your answer! After installing the packages I tried running the code:
```
var five = require("johnny-five");
var Raspi = require("raspi-io").RaspiIO;
var board = new five.Board({
  io: new Raspi()
});

board.on("ready", function() {

    console.log("Board ready, lets add light");
});
```

Which keeps running into this error. Even after rebooting. 

```
pi@raspberrypi:~/node-pixel-test $ node index.js
internal/fs/utils.js:269
    throw err;
    ^

Error: EACCES: permission denied, open '/sys/class/leds/led0/trigger'
    at Object.openSync (fs.js:462:3)
    at Object.writeFileSync (fs.js:1384:35)
    at new LED (/home/pi/node-pixel-test/node_modules/raspi-led/dist/index.js:37:18)
    at Object.createLED (/home/pi/node-pixel-test/node_modules/raspi-led/dist/index.js:62:16)
    at new LEDManager (/home/pi/node-pixel-test/node_modules/j5-io/dist/managers/led.js:32:30)
    at new J5IO (/home/pi/node-pixel-test/node_modules/j5-io/dist/index.js:116:32)
    at new RaspiIO (/home/pi/node-pixel-test/node_modules/raspi-io/dist/index.js:122:12)
    at Object.<anonymous> (/home/pi/node-pixel-test/index.js:5:7)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10) {
  errno: -13,
  syscall: 'open',
  code: 'EACCES',
  path: '/sys/class/leds/led0/trigger'
}
```

Do you have an idea what could cause this?

ajfisher, 2022-01-14:
Okay this is most likely a permissions issue of the user running the script
versus permissions available on the device.

You can check if this is the case by simply running your script as a super
user - ie `sudo node script.js`

If that resolves it then I think there's a way you can solve this by adding
your user to a specific group but I"m not sure off the top of my head which
one. `dialout` is what you usually use for serial USB devices but not sure
if that will work for I2C bus. There may be some more info on the raspi-io
repo. I'm on phone at the moment so will have a look when I get back to my
laptop and can dig around a bit more.

aronvandervalk, 2022-01-15:
Thanks for your answer. The `sudo node index.js` did the trick!

Thanks for helping me out! I appreciate it very, very much. Now I can start tinkering with your library and implementing some light installation ideas. Nice :-)

ajfisher, 2022-01-15:
Glad to hear that worked ok.

Please share when you do - love seeing what people come up with.

closed 2022-02-21 by ajfisher